martin-ger/esp_wifi_repeater #3: Max Stations

capedra, 2017-01-11:
I suggest you to let the user configure the maximum number of stations using telnet and also the option to set the ESP to work at 160 MHz. Would also be great to see it working on Wemos D1 Mini with battery shield. With my tests, I got 10 Mbps on the ESP-01.  Congratulations for this amazing firmware.

martin-ger, 2017-01-11:
The max nuber of stations is limited by the Espressif libs to 8 anyway. Thus, a function may only further reduce this limit. Would this really be helpful faeture?

Will experiment a little bit with overclocking...

capedra, 2017-01-12:
Good to hear from you, Martin. According to some users from ESP8266.com, the max number of stations can be changed directly on Espressif libs but I don't know yet what is the outcome if you change it. I think it's helpful for places where there are more than 8 devices around and you don't want to use another repeater. I will try to test your great project using Wemos D1 Mini soon. Thanks!

capedra, 2017-01-13:
Would be great to configure the desired CPU speed using Telnet. Thanks!

martin-ger, 2017-01-13:
Try the new version...

martin-ger, 2017-01-14:
Espressif says: limit of maxstations is 8 in the latest SDK: http://bbs.espressif.com/viewtopic.php?t=1894
This corresponds to my tests: setting it to any value greater than 8 in the config will still respond 8 when you read it back. Currently , I don't see a chance to increase this.

closed 2017-01-14 by martin-ger