martin-ger/esp_wifi_repeater #6: CMD>exit to close serial conection

RibeiroSantos, 2017-01-13:
CMD>quit or CMD>exit to close the serial connection? Thanks,

martin-ger, 2017-01-14:
What do you mean with "closing the serial connection"? Disabling it (until factory reset)? What is the intention? Nobdy can do any config changes any more? I think "lock" does pretty much the same thing.

capedra, 2017-01-15:
In my humble opinion, he meant that you could type "exit" or "quit" to terminate a remote session when using telnet.

martin-ger, 2017-01-15:
"quit" in a telnet session already works, I could also terminate the serial session with "quit". This would meant, ist stops listening on the serial port. But how schould one be able to start a new session? Never? Never before reboot?

capedra, 2017-01-15:
Hello Martin,

I think "stealth" would be an alternative to "exit", because you could stop
echoing anything from serial, and/or you could change the GPIO number of TX
pin (I don't think it's a good idea), and/or stop blinking this very bright
LED on my face (haha), and/or stop listening on port 7777, but still
working as a NAT Router, making the module in a "stealth" mode, not letting
the admin change any configuration anymore, forcing him/her to re-flash the
device in the future if he/she wants a new configuration.

RibeiroSantos, 2017-01-19:
I use putty, and my request was to close the session ( serial connection) as happens using telnet. But it seems to be a software feature. Thanks

closed 2017-01-19 by RibeiroSantos