martin-ger/esp_wifi_repeater #7: NodeMCU v2 Amica

capedra, 2017-01-14:
Hello Martin, I don't know why but this project is not working at this board (https://www.seeedstudio.com/NodeMCU-v2-Lua-based-ESP8266-development-kit-p-2415.html). The LED from the ESP-12 keeps blinking all the time and I can't even debug it because the serial monitor stays busy all the time. Also, the AP MyAP does never show up.

capedra, 2017-01-14:
Ok, here it is the output, which shows that it keeps rebooting all the time:

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 29232, room 16 
tail 0
chksum 0xd1
load 0x3ffe8000, len 1964, room 8 
tail 4
chksum 0xe9
load 0x3ffe87b0, len 3256, room 4 
tail 4
chksum 0xeb
csum 0xeb
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 29232, room 16 
tail 0
chksum 0xd1
load 0x3ffe8000, len 1964, room 8 
tail 4
chksum 0xe9
load 0x3ffe87b0, len 3256, room 4 
tail 4
chksum 0xeb
csum 0xeb
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 29232, room 16 
tail 0
chksum 0xd1
load 0x3ffe8000, len 1964, room 8 
tail 4
chksum 0xe9
load 0x3ffe87b0, len 3256, room 4 
tail 4
chksum 0xeb
csum 0xeb
rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 29232, room 16 
tail 0
chksum 0xd1
load 0x3ffe8000, len 1964, room 8 
tail 4
chksum 0xe9
load 0x3ffe87b0, len 3256, room 4 
tail 4
chksum 0xeb
csum 0xeb
rf_cal[0] !=0x05,is 0xFF

capedra, 2017-01-14:
The exact same problem happens to the Wemos D1 Mini. On both NodeMCUs I tried flashing from your firmware, because I don't know how to build from scratch. Maybe if you could explain how to build the binaries step by step on Youtube using Windows/Linux, it would be really helpful. Thanks a lot! I hope I've been helping you a lot lately.

martin-ger, 2017-01-14:
Obviously, your board is rebooting all the time without really starting the repeater firmware at all. According to http://www.esp8266.com/viewtopic.php?p=2096#p2112 "rst cause:2" means reset pin is pulled. Are you pressing the button?

Just tested it: for me the lastest firmware downloads, boots, and works fine on a Wemos D1 Mini. Don't have a NodeMCU at hand, but if should be fairly compatible to the Wemos.

The download procedure on Linux is just the following:
1) Make sure you have the esptool.py on your PATH
2) Stop any serial terminal program that might use the serial port
3) connect the Wemos via USB to the Computer
4) "cd" to the directory where you have the firmware binaries
5) sudo esptool.py --port /dev/ttyUSB0 write_flash 0x00000 0x00000.bin 0x10000 0x10000.bin
   (no button or any electical signal on any GPIO port required)
6) Wemos starts directly after the download finished - just to be sure, press reset once
7) connect to the serial port via a terminal program. I use "minicom -D /dev/ttyUSB0 -b 115200"
8) You should see some debug output and the serial console...

capedra, 2017-01-14:
> viewtopic.php?p=2096#p2112 "rst cause:2" means reset pin is pulled. Are

I am not pressing the reset button. Are you connecting the TX/RX of Wemos
after installing the firmware or just using 5V/3.3V VIN? My device is a
Wemos D1 Mini v1. Is your Wemos v1 or v2? Because some pins change from one
version to another.

capedra, 2017-01-14:
I meant that I don't know how to build the binaries from the ESP SDK and your other library. So sorry because I was so tired that I forgot to include the params "-fs 32m -ff 80m" to esptool. Will test again as soon as possible with Amica. Thanks!

martin-ger, 2017-01-14:
This is my Wemos : http://www.ebay.de/itm/272271662681 I guess it is compatible to V2.
I connect nothing but USB for downloading and testing. USB provides power supply and serial link.

capedra, 2017-01-15:
Okay, thank you very much. It's working on NodeMCU v2 Amica with the command that you sent to me:

esptool.py --port /dev/ttyUSB0 write_flash -fs 32m -ff 80m 0x00000 0x00000.bin 0x10000 0x10000.bin

Thank you very much!

capedra, 2017-01-15:
I'm sending this message right now with the help from your outstanding project, which is working with a NodeMCU Amica v2 which is connected to an ESP-01, which is connected to a Linksys Repeater, which is connected to a Linksys Main AP, which is connected to a D-Link Modem, which is connected to the WWW, which has its own creator Sir Tim Berners-Lee.

martin-ger, 2017-01-16:
That sounds great - obviously, it works... :-)

closed 2017-01-16 by martin-ger

yashp96, 2018-08-13:
Please help me out. I have two node mcu and 3 ESP01. I was able to upload and run the firmware successfully in one of the nodemcu but the other one shows error when connected to COM port shows chksum errors. Also the firmware fails to upload on all 3 ESP-01. I am posting the screenshot for NodeMCU. 
[image]