martin-ger/esp_wifi_repeater #8: Help Installing

RibeiroSantos, 2017-01-16:
Hi,
I'm trying to compile your code but having problems. Are you compiling with Windows ou Linux?

I follow the instrutions in https://github.com/pfalcon/esp-open-sdk , and build the blinky example with sucess (after added -lc in LDLIBS)

When you said 

> Drop the esp-open-lwip into the esp-open-sdk, i.e. copy it into the build tree, and build the esp-open-sdk

 you mean: 
- rename the original esp-open-lwip directory to other name 
- execute `git clone https://github.com/martin-ger/esp_wifi_repeater.git` in esp-open-sdk directory 
- and execute "make rebuild"?

Thanks,
Ribeiro Santos

martin-ger, 2017-01-16:
That's right. However, make sure that you are on my "nat"-Branch of esp-open-lwip. The easiest way to guarantee that is to download:
https://github.com/martin-ger/esp-open-lwip/archive/nat.zip
unpack and replace that in the esp-open-sdk tree. "make clean" in the esp_open_lwip dir and once again a "make" in the upper esp_open_sdk will do the job. This installs a new version of the liblwip_open.a that contains the NAT-features

(Recently I had an issue with tcp.c (tcp_sane())- maybe these should be one endif TCP_DEBUG one proc further in the code) 

I am building on Linux.

RibeiroSantos, 2017-01-16:
Sorry again ask, but, you have esp-open-sdk installed at `/opt/Espressif/esp-open-sdk` or just `/opt/esp-open-sdk`?
Thanks a lot

martin-ger, 2017-01-16:
Acutally, I have it at /home/martin/github/esp_open_sdk. Doesn't matter, as long as you adapt the pathes in esp_open_lwip/Makefile in the first few lines.

RibeiroSantos, 2017-01-17:
>  I had an issue with tcp.c (tcp_sane()

Same problem, can you help how you fixed?
Thanks

martin-ger, 2017-01-17:
Just comment it out - it is not needed...

RibeiroSantos, 2017-01-17:
Now I get

```
lwip/core/tcp.c:1664:19: error: macro "tcp_pcbs_sane" passed 1 arguments, but takes just 0
 tcp_pcbs_sane(void)
                   ^
lwip/core/tcp.c:1665:1: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
 {
 ^
```

martin-ger, 2017-01-17:
Did a commit on tcp.c in the nat branch - should work now...

RibeiroSantos, 2017-01-17:
Worked (with Ubuntu Beta of Windows 10) !!! 
Thanks a lot!! 
When you come to Lisbon, I offer you a beer! ;)

closed 2017-01-17 by RibeiroSantos