martin-ger/esp_wifi_repeater #9: About https://github.com/daiyinger/esp-open-lwip/tree/nat

daiyinger, 2017-01-18:
i got liblwip_open.a after build esp-open-lwip porject.
can i use lwip_open.a to replace the  liblwip.a file of the lib directory in SDK(like ESP8266_NONOS_SDK_V1.5.4_16_05_20)  to implement the NAT function.

universam1, 2017-01-18:
I would eagerly interested to make use of this great work with Arduino environment (because of libraries etc.) and wonder if this would be a way to port it?
Any other suggestions?

martin-ger, 2017-01-18:
Yes, liblwip_open.a is meant to be a complete replacement for liblwip.a. Without enabling the NAT-funktionality it acts like the normal liblwip.a, but it already does forwarding between interfaces.

To enable NAT on the SoftAPs interface, simply set the new "napt"-flag in the netif structure of the APs interface. (the AP interface is the one with netif->num == 1). 

In the DHCP server the lib add a function to set the distributed DNS server address (originaly it was hardcoded to sed the address of the SoftAP):
`void dhcps_set_DNS(struct ip_addr *dns_ip;);
`
I already added the esp_conn-files required for the TCP- and UDP-compatibility to liblwip. You might also want to include app/mdns_conn.o, app/netio.o and app/ping.o in the Makefile.open to make it complete (didn't need that for the repeater).

martin-ger, 2017-01-18:
I have no understanding of the internal structure of the Arduino environment. I would guess, it is just a replacement of the lib and a handful of new API-fuctions. Perhaps anybody out there has more experience with pimping the Arduino-IDE...?

capedra, 2017-01-19:
I could try porting it to Arduino IDE if I can get the source code and the working flow of the code. I think the best board for this repeater is the Wemos D1 R2 because you have lots of GPIOs available and it solves the problem of power supply when you don't have a USB port available around there. Also I've been testing the Amica but the ESP-01 has been working a lot better than it, because the Amica has been stopping repeating the signal from time to time and you can see lots of errors on ping.

capedra, 2017-01-19:
Besides that, I don't think it's a problem with the Amica's power supply because I have been using a 5V 1A wall adapter.

martin-ger, 2017-01-19:
What exactly do you need? I think, the first and most important step would be, to integrate the patched liblwip-open.a in the Arduino IDE. Then, the basic repeater functionality is just setting up the STA and the SoftAP and switching on the NAT. This should be a pretty short sketch...

The remaing fuctionality of the repeater software (mainly the console interface) is more or lees subject to change: a web-interface would be perhaps even nicer. 

The remote monitoring feature requires acces to the netif-strucure to redirect the input/output functions. Should be possible under Arduino as well, but I guess most people are even not really interested in that.

Concernig the problems with the Amica: what does it look like, when it stops repeating? Resets? Other error messages? Just nothing? Does it recover?

A problem when repeating is buffer space. Under high load the lwip-stack runs out of buffers an this results in a drop of packets. TCP compensates this with retransmissions, but the user experiences a sudden drop in the transfer rate because of TCPs contention handling and slow start. More buffers and perhaps an increased clock-rate of 160MHz may help a little bit (havn't tested that so far). 

From the logic of the program there is no difference between the various flavors of ESP-boards and the core MCU and the memory is the same on all boards. Maybe Espressifs ROM is somewhat different (anybody out there with knowledge about that?). The difference in SPI flash size should not matter once the program is running, as it does not use flash after reading its config.

capedra, 2017-01-19:
Hello my friend. About the Amica, it keeps blinking the LED slowly after
some time connected and then it looks like it loses the connection with the
main AP, but after some seconds or a minute it comes back working fine, but
it doesn't reset or anything because in the meantime I don't get
disconnected from the SSID of its own.. This problem happens very often
that I decided to give up using Amica and only use the ESP-01 at 160 MHz
connected to an Arduino Uno that serves as a FTDI. Of course that this
problem of using the Arduino Uno as a power supply gets solved when I use
the Wemos D1 R2 but I just have the D1 R1 which I'm using in another
project.

About what I need, I want to know the working flow of the program, like
where the setup and loop functions would fit on it. Since you made a
firmware, I have to study the source code to develop for Arduino.

The remote monitoring function could be integrated with the web interface,
because a visual approach is a lot easier to non-tech people to deal with.

Finally I think that this buffer space issue could be solved by using the
lots of storage from flash because it's very fast, unusable and stable
(I've been testing the flash for almost a year on Amica).

universam1, 2017-01-19:
> What exactly do you need? I think, the first and most important step would be, to integrate the patched liblwip-open.a in the Arduino IDE. Then, the basic repeater functionality is just setting up the STA and the SoftAP and switching on the NAT. This should be a pretty short sketch...

Thanks for explaining this, I wonder could you highlight the calls a bit clearer where and how this switching on the NAT is to be done?
I understand that the Arduino core could be patched with your `liblwip.a` which should be sufficient for basic NAT functionality?

martin-ger, 2017-01-19:
The liblwip_open.a in the "nat-branch" implements the new calls introduced by NeoCat here: https://github.com/NeoCat/esp8266-Arduino/commit/4108c8dbced7769c75bcbb9ed880f1d3f178bcbe and also a function "void dhcps_set_DNS(struct ip_addr *dns_ip;)" to set the DNS address that is distributed via DHCP.

In the next days I will upload a version of lwip with a better documented API.

martin-ger, 2017-01-21:
Added the interface in a little bit cleaner way in https://github.com/martin-ger/esp-open-lwip

closed 2017-01-30 by martin-ger