martin-ger/esp_wifi_repeater #10: It is possible to filter/change requests?

RibeiroSantos, 2017-01-19:
Hi,
As far I understand, all client comunications goes by the repeater (this case the esp8266), right? 
My question is if it is possible, to change or filter some requests.

By example, a client A sends a message to the esp (the repeater) if the message has some predefined format, it musts be send to all other clients, and changes the response to let client A knows, his message, was well received. It is possible? Or at least in http requests? 

My goal is to do a kind of MQTT Broker, which seems possible looking at your great code. Maybe not to handle 1000 clients but if may handle 8, I will be happy :)

Thanks a lot,

martin-ger, 2017-01-19:
Yes, it would be possible to filter or even change messages that go through the repeater (router). Actually, it is already done in the lwip stack when rewriting network addresses and port numbers while doing the NAT. In the functions "my_input_ap()" and "my_output_ap()" the code has access to all packets going in and out. This is, how the monitoring is done. Rewriting packets is possible, however the checksums and in TCP the SYN and ACK numbers have to be adjusted. Also, when packets become a different size, possibly some handling for the pbufs is required. This can be done, but this is more hacking (the NSA-style) than a way of implementing a network service.

If you want to implement a MQTT broker on the repeater, why not using a "normal" TCP-service.
Would work similarly as the console service right now and can be integrated with the router functionality. If code size becomes an issue, one can easily drop the monitoring for this version.

RibeiroSantos, 2017-01-19:
> why not using a "normal" TCP-service

 Sorry, but don't understand what you mean... Could you help me? Or with some kind of pseudo code...
Thanks again!

martin-ger, 2017-01-19:
Well, here: https://github.com/Spritetm/esphttpd for example people have set up a complete web server on an esp. In my code there is the telnet console on port 7777 a native TCP server (service) on the esp. An MQTT broker works similar (from the network point of view). There is somebody out there claiming to have code for a small MQTT broker on an esp (https://myesp8266.blogspot.de/2016/03/mqtt-broker-on-esp8266-2.html) - however he's not willing to share that...

closed 2017-01-21 by martin-ger