martin-ger/esp_wifi_repeater #14: Web Interface

martin-ger, 2017-02-13:
Would love to see it with a web interface instead of the console. libesphttpd should be a good option to implement that. Adding the server structure shouldn't be a big deal. For me a problem would be the page design and the JS-part. I am really not into that. Anybody willing and able to support this?

MarsYong, 2017-02-16:
Use a period of time after the "NAT table full", what may be the cause of it？

martin-ger, 2017-02-16:
This happend, when you are using a lot of connections a the same time. Currently the table can hold 512 NAPT translations simultaniously. After a connection is closed, there is still a timeout for the last FIN packets. This is a transient state and it will recover and free entries after the timeouts.

You can increase the table size or decrease the timeouts in:
 esp-open-lwip/include/lwip/lwip_napt.h

MarsYong, 2017-02-16:
Thank you for your help, the way to solve the problem is this, but I still do not understand "IP_NAPT_DYNAMIC = 1", what is the function？What problem can be solved by it？
 And I set it , it will always appear "NAT table full". Why?
“LmacRxBlk:1” is also often appear，why？

martin-ger, 2017-02-16:
With "IP_NAPT_DYNAMIC = 1" the memory for the NAPT tables is allocated only, if "ip_napt_init(max_nat, max_portmap)" is called explicitly before enabling NAPT. This saves memory, if you use my version of open-lwip but don't need NAPT. The tables still have fixed size after initialization. If you set this option without calling  ip_napt_init() in your program, the tbales have null size -> always full.

"LmacRxBlk" comes from Espressif's WiFi code and as far as I understand inidicates that currently there are no more buffer availablle. Leads to drop of single packets,

piyotoru, 2017-02-17:
Hi! For example I got a problem with connecting to ssid name ex. POWER NET not POWERNET

martin-ger, 2017-02-17:
set ssid POWER\ NET
should work

piyotoru, 2017-02-19:
Thanks. works!

timpur, 2017-02-22:
I think it would be nice to see a Web Interface.

closed 2017-03-01 by martin-ger