martin-ger/esp_wifi_repeater #15: Repeater Stops Working after Awhile

timpur, 2017-02-19:
I've noticed after some time the repeater stopes working and clients can't send data through.

Admittedly I haven't updated the Repeaters software in some time so I'll give that a try soon.

Just wondering if anyone else has had this problem ?

Will try to diagnose the actuall problem soon. Just a open question. Don't know what's really going on with this issue. But I do know a simple power off and on fixes the issue :P. Thinking maybe adding a restart at midnight or some time specified could be nice to have as a last resort thing ?

Sorry for the non help full information.

martin-ger, 2017-02-19:
For debugging purpose try to connect a serial terminal and log the output .

timpur, 2017-02-21:
Yeah to be honest it could be that. Now that ive been playing around with that ESP module ive noticed that the flash mem is playing up so i think its the actual module. Im going to replace with a new module and see  how that goes.

timpur, 2017-02-21:
Also just like to say this is an awesome project and i think its amazing you can turn a $3 device into a wifi repeater that does the same work as something more like $50. Its grate for extending your wifi network to support small Devices like IOT devices alover your house and beyond into the garden :P. Grate work and thanks for your dedication. Love it !!!

closed 2017-02-22 by timpur