martin-ger/esp_wifi_repeater #18: Error after entering 'make'

nisc3, 2017-02-26:
I'm getting this error when I run make command 
`Makefile:181: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
Makefile:176: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
make -C crosstool-NG -f ../Makefile _ct-ng
make[1]: Entering directory '/opt/esp-open-sdk/crosstool-NG'
../Makefile:181: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
../Makefile:176: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
./bootstrap
Running autoconf...
./bootstrap: 5: ./bootstrap: autoconf: not found
../Makefile:143: recipe for target '_ct-ng' failed
make[1]: *** [_ct-ng] Error 127
make[1]: Leaving directory '/opt/esp-open-sdk/crosstool-NG'
Makefile:140: recipe for target 'crosstool-NG/ct-ng' failed
make: *** [crosstool-NG/ct-ng] Error 2
`
I'm using Rpi3

martin-ger, 2017-02-26:
You are getting this when you try to build the esp-open-sdk? You are following the instructions on https://github.com/pfalcon/esp-open-sdk ?

On which OS? I did all my builds on Linux, but the project uses the standard makefile, so there should be nothing specific here? Can you build any other ESP sample project?

nisc3, 2017-02-28:
I'm running Raspbian Jessie on rpi3. I was able to compile the esptool SDK 2.0.0 for ARM which I'm using now

martin-ger, 2017-02-28:
Can you now replace the esp-lwip-open tree with my code from here https://github.com/martin-ger/esp-open-lwip and do the build again?

nisc3, 2017-03-01:
I removed the entire contents from that folder and added yours, still no luck...

nisc3, 2017-03-01:
I made it work by flashing firmwares from a windows PC, had a lot of problems when tried to compile with raspberry pi

closed 2017-03-01 by nisc3

martin-ger, 2017-03-01:
I am just building the esp-open-sdk on a raspi - it takes... "forever". Maybe it is completed tomorrow. Then I will look into it, where there might be problems....

martin-ger, 2017-03-02:
After 16h of building the esp-open-sdk and especially the gcc on my Raspi B it failed with some unclear error during linking of phase 1 of gcc (which is even half way of building the SDK), It seems to me that using a Raspi it not a verry good idea and I am not able to further dig into that.

If you only have a windows machine, a VirtualBox with a minimal Ubuntu is perhaps a better choice - this worked for me...