martin-ger/esp_wifi_repeater #19: Compile with esp-open-RTOS, any help?

RibeiroSantos, 2017-02-27:
Hi, I'd like to compile with esp-open-rtos  but don't know were to star. Compiling it reports many errors so  it shouldn't be easy, at least, to me. Any help were to star? 

Thanks

martin-ger, 2017-02-27:
First step would be to integrate my the esp-open-lwip version in the build environment. Should be just a replace of the lib and the includes. This already does the main job of forwarding and NAT-ing. If you set up the ESP in StationAP mode, mark the AP interface as NATed, and connect the STA it should work independantly of any additional program code. The current user_main.c is only for the hole admin stuff on configuring the router (and the monitoring).

Btw, why would you like to have it on RTOS?

RibeiroSantos, 2017-02-27:
I made a custom board with one  ESP and a PIC. Some of the specifications are: control 2 AC power channels (with independent dimmer), 2 inputs, 2 ADC, zero cross detection, I2C, SPI,... and after some searchs, at the moment I consider the RTOS the best choice. 
With RTOS using the tasks I think it will be more easy the development.

martin-ger, 2018-09-15:
No, I don't have plans to do so - no time. Would be happy, if you start an attempt...

gr01d, 2018-09-15:
@martin-ger will try.

colesnicov, 2020-05-20:
> @martin-ger will try.

Hello. Progress?

gr01d, 2020-06-03:
Haven't started yet, I just can't find good documentation/information about the esp_open_rtos.

colesnicov, 2020-06-03:
Why not for ESP8266-RTOS-SDK?

gr01d, 2020-06-03:
Proprietary license, I plan to use the esp_open_rtos version.