martin-ger/esp_wifi_repeater #20: LoRa ?

laszlo-kocsis, 2017-02-27:
Hi Martin,
I've tried your repeater mod and it works very well. Now I'm wondering if this could be extended to really great distances with two esp8266 connected to a LoRa modules (eg. SX127x based ones or similar SPI bus based LoRa devices). That would be truly amazing!

martin-ger, 2017-02-27:
Might work and I have actually ordered two of these LoRa modules already. Technically it might work to treat the LoRa-Link as a kind of a p2p link between two routers. However, the expected bandwidth would be veeeeery slow and IP is because of its overhead surely not the best protocol for this technology. Also putting heavy load on an private LoRa link would violate the fair use policy, as it would interfere with anybody else trying to use LoRa in the area.

laszlo-kocsis, 2017-02-27:
Yes, this would be very slow and I don't think it is for video streaming :). I think the implementation could be a reference for any serial type link in the future. LoRa can be a "good" experimental platform since this is the most problematic one. As for the speed: in this case esp8266 would compress raw wifi frame before packetizing for LoRa sending... and other tricks can be used to "speed it up" a bit.

dpwallac, 2017-03-27:
LoRa would be great, and not specifically with LoRa WAN either, the P2P would suffice for general remote data collection. I connected 4 Wemos in series (1 ESP communicates to the next as its AP) using Martins code and connected a Linux device at each level. The Linux machine could browse the web using level 1-3, but level 4 failed... most likely due to heavy delay in relay between each ESP.

How does this relate to LoRa? Sending simple messages across the same setup, say PING to bbc.com, did not seem to suffer a degrading packet loss or increase in round trip time, being 16012ms @ 5% packet loss at each level (not degrading at each level!). So if you only need simple communication, like small IoT MQTT messages, the ESP8266 would be fine in a series (or specifically Star) configuration (say 4 layers at 4096 devices (8^4) in theory). LoRa would be great at the same Use Case, small messages but over longer distance. Would defiantly lower development overheads being able to use HTTP or TCP/IP libraries instead of bespoke radio packet assembly. I would love to test a LoRa version of this!

Great work Martin!!

pat2man, 2017-12-26:
As @martin-ger said IP is probably not the best protocol over LoRa since it could easily cause unnecessary packets to be sent over a public channel that has limited capacity.

Instead it would be interesting to bridge MQTT over LoRa. This would allow two mesh networks to communicate over LoRa via MQTT. Since MQTT is designed for low bandwidth lossy connections it would match up better with the requirements of LoRa. (You could even use MQTT-SN which further reduces bandwidth)

martin-ger, 2017-12-27:
That's right. Did some experiments with LoRa. It is incredably slow and packets are short compared to WiFi (that's the spec). Doesn't make any sense to do TCP/IP over it. LoRaWAN is a reasonable approach for integrating it into the IoT, MQTT-SN sounds good as well, if you want a pure local setup. Maybe I'll find a student willing to setup a prototype for that...

closed 2017-12-31 by martin-ger