martin-ger/esp_wifi_repeater #21: Select Phy Mode ?

timpur, 2017-02-28:
Wondering if its possible to select which phy mode the repeater is in ?

martin-ger, 2017-02-28:
Not yet at the interface - what do you think is needed?

timpur, 2017-03-01:
Just to pick between the 3 phy modes for wifi 
B,G,N
Best out come would be to be able to pick which phy the Station is in and which mode the AP is in, but not sure if the ESP allows 2 different phy modes so atlest be able to pick which phy mode for both :) 
ATM I believe it's using G. I want to try different modes to see which is best result for speed and distance for different situations.

Thanks.

martin-ger, 2017-03-02:
Made some experiments with different PHY_MODEs:
Standard setting is 'n', however the SDK documentation says: "_Note: ESP8266 soft-AP only support bg._", So, I assume it actually works only in 'g' mode when in StationAP mode as it is done in the repeater. There are no distinct settings for STA and AP. 
At a first look, mode 'g' works exactly the same as 'n'. 
When set to 'b', it immediately loses connection to my home router and cannot reestablish. :-(

timpur, 2017-03-07:
oh didnt know that AP is only bg.

thats odd sounds like your router doesnt support b ?

i still think as a feature point of view, it woild be nice to be able to select these modes. This way we can tweak to get the best performance or distance out of this repeater. If its not to complicated to implement?

martin-ger, 2017-03-07:
Made branch "phy_mode" that includes a includes a command "set phy_mode [1|2|3]". We might try, if it has any useful effect. Currently I don't think it should be in the master branch...

timpur, 2017-03-12:
Whats wrong with providing extra configuration?

closed 2017-05-19 by martin-ger