martin-ger/esp_wifi_repeater #22: Web interface to control the repeater

nisc3, 2017-03-01:
Is it possible to build a web interface to add/remove networks, to change the name and password of the extended networks etc.?

martin-ger, 2017-03-01:
As I said in another issue: certainly possible, even desirable. However, I am not a Web/GUI person: libesphttpd should be a good option to implement a Web-GUI. Adding the server structure shouldn't be a big deal. For me a problem would be the page design and the JS-part. I am really not into that. Anybody willing and able to support this?

RibeiroSantos, 2017-03-01:
I will try to design some pages using json and ajax (but busy until next week)

martin-ger, 2017-03-01:
Would be great. I could definitely add the server structure.

closed 2017-03-06 by nisc3