martin-ger/esp_wifi_repeater #23: No AP after flashing

anudeepND, 2017-03-08:
I flashed two .bin files using esptool.py. But there is no AP after flashing, serial monitor shows gibberish texts and symbols and it dosen't accepts any command. I'm using the latest SDK

martin-ger, 2017-03-08:
Do you use an ESP-01 module? If so, try SDK 1.5.4 instead of 2.0.0 (perhaps test with my precompiled firmware_1.5.4 first). Guess it has something to do with the smaller flash and the different linker map, but I havn't understood this completely...

anudeepND, 2017-03-08:
Correct me if I'm wrong, I have esp8266 SDK 2.3 installed on my system. I have downloaded the repo and flashed the firmware using esptool.py. Is this the correct way of doing?

martin-ger, 2017-03-08:
esp8266 SDK 2.3 is the version number of the ESP Arduino environment. My projekt ist based on the esp-open-sdk (https://github.com/pfalcon/esp-open-sdk ) - something slightly different. This can be installed with Espressif's SDK 2.0.0 (default) or SDK 1.5.4 (somewhat older) (https://espressif.com/en/support/download/sdks-demos ). It compiles on bot envirnments, however, as said above, some ESP-01 modules fail with the firmware build with the SDK 2.0.0, SDK 1.5.4 (0x00000.bin and 0x40000.bin) works.

anudeepND, 2017-03-08:
So I need to compile Open-sdk and then flash the firmware?

martin-ger, 2017-03-08:
To compile the firmware yourself - yes. If you you only want to flash the provided binaries, you can use any ESP flashing tool like the one referenced in the Readme.
You still havn't answered: do you use an ESP-01 Module? (then try it with the files provided in the directory firmware_1.5.4)

anudeepND, 2017-03-08:
Yes I'm using ESP-01 module. I have downloaded open-sdk and replaced the file lwip. Then?

anudeepND, 2017-03-08:
*esp-open-lwip

martin-ger, 2017-03-08:
Adapt the variable to "VENDOR_SDK = 1.5.4" in the esp-open-sdk's Makefile and build the esp-open-sdk like explained here: https://github.com/pfalcon/esp-open-sdk#building
"make" downloads some additional stuff and builds the toolchain... (takes some time)
Then replace the esp-open-lwip in the tree and do the esp-open-sdk build again ("make") just recompiles the lwip...
Then you can download the esp_wifi_repeater in any directory you like, adapt the pathes in the Makefile, build and flash it.

anudeepND, 2017-03-08:
Flashing was successful, I can connect to MyAP. I opened serial monitor and I can't set ap_ssid.
My WiFi name is prime...

martin-ger, 2017-03-08:
Try: `set ap_ssid My\ WiFi` (blackslash quotes the blank)

anudeepND, 2017-03-08:
And telnet says 'no route to host'. I scanned MyAP network with network scanner and it shows no services are available. Telnet disabled?

martin-ger, 2017-03-08:
telnet <host-ip> 7777

anudeepND, 2017-03-08:
I did set ap_ssid wifiname and save, then pressed rst button. But name is still MyAP

martin-ger, 2017-03-08:
What does `show` say?
Set the params, save, and then show once again (just to make sure)...
Then `reset` - I am not aware that this ever failed...

martin-ger, 2017-03-08:
Btw: _Maker Saga_ wrote independently an instruction on setting up the repeater: http://www.instructables.com/id/Cheap-WiFi-Range-Extender-for-IoT/?ALLSTEPS

anudeepND, 2017-03-08:
Serial monitor displays no ssid found. After setting parameters, it is still showing no ssid found.I changed the WiFi name itself to ssid and it successful connects.

anudeepND, 2017-03-08:
Still it's not saving any commands, I will try with another module.

closed 2017-03-08 by anudeepND

martin-ger, 2017-03-08:
Please check power supply! Flashing with `save` might fail with not enough current...
If you have compiled it, you might try to change the values already in user_config.h

anudeepND, 2017-03-09:
Is it possible to make it work with a simple arduino code?

martin-ger, 2017-03-09:
It is probably not too hard to integrate my lwip-version into the Arduino environment, but I am not into that - sorry...

anudeepND, 2017-03-09:
SDK compiled successfully, when I try to compile this repeater it gives 
`Error calling xtensa-lx106-elf-readelf, do you have Xtensa toolchain in PATH?
Makefile:115: recipe for target 'firmware/0x00000.bin' failed
make: *** [firmware/0x00000.bin] Error 1
`
I have placed esp_wifi_repeater folder inside esp-open-sdk folder and edited build-area in makefile to /home/pi
Is that correct?

anudeepND, 2017-03-09:
Finally! It's working. I'm getting 8.3mbps down and 8.2mbps up speed!! Thanks Martin

martin-ger, 2017-03-09:
Good News! Have fun...