martin-ger/esp_wifi_repeater #24: Porting ESP32

SmartArduino, 2017-03-14:
When porting ESP32

martin-ger, 2017-03-14:
Currently, I do not have plans to do so, as I don't even have an ESP32 here. 20$ for a developer board makes it somewhat unattractive for that purpose, as a simple OpenWRT platform inkluding power supply costs about the same and is much more flexible.

However, the souces are there and I assume the open_lwip lib doesn't need much porting...

ricksondpenha, 2017-04-28:
hi can you guide us to port this to esp32. I would very much like to see the results on a esp32 (performance wise)

RibeiroSantos, 2017-04-28:
The open_lwip lib isn't the same to use with esp8266 and esp32?

ricksondpenha, 2017-04-28:
awesome. keep us posted.

Derek-X-Wang, 2017-06-14:
Hi, @martin-ger.  try to port to esp32. I modified lwip in esp-idf based on your [esp-open-lwip](https://github.com/martin-ger/esp-open-lwip). The issue is all dns queries get no response, but ICMP type3 code3(port unreachable). It seems that the udp port that sent the request, was shut down before received the response. Any idea... [my modified version](https://github.com/Derek-X-Wang/esp-idf)

martin-ger, 2017-06-15:
In my version, DNS requests are sent directly to the upstream DNS server. The clients learn about it from the DHCP config. Could you check, which DNS server is distributed (and used by the clients)? It should NOT be any IP-address of the ESP, but the DNS server, that the ESP learned from its DHCP config. I set this with the call to dhcps_set_DNS() in the main program, which I have added to dhcpserver. You might  try also "8.8.8.8" (Google's DNS) here.

ricksondpenha, 2017-07-07:
hi, @martin-ger any update on the esp32 port?

Derek-X-Wang, 2017-07-07:
Don't have time to work on my esp32 project recently, so I haven't try martin's solution yet. But I think espressif does some odd things in their wifi driver which is closed source. It closes the udp port after a dns request is sent.

ricksondpenha, 2017-08-19:
hey any update on this?

ricksondpenha, 2017-10-05:
Hello... Any plans for porting this on esp32? It would be a great feature to have plus the performance!!

WayneKeenan, 2018-08-08:
I wondered if the port to ESP32 is still on the 'TODO' list ?
Thanks 
Wayne

martin-ger, 2018-08-09:
Actually, I would love to do so (especially porting the lwip NAT, as Espressif announed to include that but hasn't done that to my knowlegde up to now), but currently, I don't find the time...

Feel free to support (do) this...

martin-ger, 2020-02-17:
Find a very basic version here: https://github.com/martin-ger/esp32_nat_router
It does the basic NAT and can be configured with serial console (Putty or GtkTerm with 115200 bps). Use the "set_sta" and the "set_ap" command to configure the WiFi settings. Enter the help command get a full list of all available commands.

closed 2020-02-17 by martin-ger