martin-ger/esp_wifi_repeater #25: Limit speed?

RibeiroSantos, 2017-03-27:
Hi, any chance to be able limit the transmission speed? 

(Building the new house and not having time, yet, to the web interface, sorry)
Thanks

martin-ger, 2017-03-29:
What do you think about? Per station? For the uplink? Would be possible to add a scheduler (leaky bucket/token bucket) to either the uplink and/or the AP. But what would be the reason to do so?

jitsuslap, 2017-04-09:
@martin-ger  I am really sorry to write it here where it's irrelevant but I really wanted to ask this. How much time did it take for you to write the code for esp_wifi_repeater; I have been trying to write my own mesh code for past two months but this seems superior. By controlling subnetting I guess I can have a whole mesh network downstream. Do tell me how much time it took for you to write this.

martin-ger, 2017-04-09:
Before using the NAT routing approach I tried to get it running as a real layer 2 repeater. After about one week of part-time work I gave up, as I found no way to send frames with src-address other than the own MAC (guess that is hardcoded in the driver). When I came across Neocat's NAT for open-lwip it took me about another week to get the basic functionalty running. Everything else since then, was additional features (monitoring, MQTT, ACLs) and CLI.

I still have an idea for an automatic mesh: upstream "mesh" (at least chaining of a number of repeaters) is not a problem once configured. However, up to now it requries manual configuration: IP subnets have to be different, at least in connected subnet, and the SSIDs as well. You can use the same SSID for uplink and downlink, but with a number of meshed repeaters with automatic uplink connect, this might end up in a loop without real connectivity to the uplink AP. 

My idea was: all repeaters have the same SSID/PWD. Each repeaters sends out additional beacons, that contain the BSSID of the repeater and the distance in hops to the uplink AP, when connected. Before a repeater connects to another AP it scans all the available beacons an decides to which it connects (trade-off link quality/level).

jitsuslap, 2017-04-09:
Making a custom probe request for the mesh would be a clean and proper solution. It would enable mesh parameters to be communicated to a new node wishing to join and make the SSID's cleaner.

I have currently made a protocol using esp-touch that makes all the esp8266's in the range of the router to connect to it through broadcasted password and automatically elect a portal device, then the rest disconnect from the router and connect to the portal device. This way you don't have to load any router or mesh configuration into the esp8266 while flashing. I have made a basic mesh that works in this instance but just now I discovered IP_FORWARD in lwip which makes my current mesh approach inferior and I wish to go NAT way.

I don't understand how can esp8266 give such great speed as wifi repeater? Over a tcp connection between esp8266 as STA and my PC (through a router) I get speed in kbps range. What's the deal?

martin-ger, 2017-04-09:
A performance issue when using TCP on the ESP might be the way the lwip stack on the ESP handles TCP. The next packet must be only send in the CB of the prevoius. Maybe this results in a suboptimal flow control.

The processor with 32 bit and 80/160MHz is fast enough to handle IP packets quickly and pass them through. There is no special trick, but with IP forwarding there is not much protocol status und processing involved.

jitsuslap, 2017-04-09:
Makes sense. Thanks!

martin-ger, 2017-04-11:
set [downstream_kbps|upstream_kbps] _value_ now limits the speed of the AP interface.

closed 2017-04-11 by martin-ger

jitsuslap, 2017-05-06:
Hey! Hope you can answer this for me.
Can the netif interface exposed in esp-open-lwip be used to send raw 802.11 packets between esp8266 devices that are not connected to each other?? I suppose devices would have to be connected as station or access-point to the mac-addr I want to send packets to using netif but I'm hoping it can be used to send packets to random esp8266 in range after reading their mac-address off beacon.

martin-ger, 2017-05-07:
No, you can't, you have to use wifi_send_pkt_freedom() for this with some limitations. Receiving then works also with limitations in sniffer mode.