martin-ger/esp_wifi_repeater #26: Feature request

adrianrb, 2017-03-28:
First, THANK YOU!
Works like a charm...

But...

1. Is it possible to module save more than one SSID/PASS (and connect to stronger signal automatically)?
2. led pwm [0..255] (it´s very strong at night in bedroom)

Adrian

martin-ger, 2017-03-29:
Will try to add the multiple SSIDs in the next "release", sounds interesting for many situations.
The PWM for the LED, well... I would prefere a HW-solution with a resistor ;-) If there is really time spare...

anudeepND, 2017-03-30:
An option to set DNS server would be nice

martin-ger, 2017-04-04:
Static DNS derver and also static IP address oft the AP are now supported.

maxboar, 2017-04-26:
Good day, very nice repeater.
Could there be an option to set in DHCP several fixed ip addresses by MAC in the internal network?

martin-ger, 2017-04-26:
You already have the option to do "save dhcp" which save the current state of the dhcp mapping MAC<->IP. Explicit definitions are curently not supported... No big deal, but requires an additional MAC-parsing function.

maxboar, 2017-04-26:
Configurable DHCP would be a great asset in future development. I am using ESP to synchronize several NanoPi Airs, and at least one of them (master node) need unchanging IP address. I'll manage somehow with "save dhcp" option, and 'll continue testing//
I am pretty newbie to programming ESP, but such capabilities presented in this branch are already delivering me some "ants in da pants" feeling =) 
MQTT broker would be also great asset. Found some goodies here https://github.com/xDWart/MQTTbroker
But it's still raw&buggy

martin-ger, 2017-04-27:
Actually, there is a development branch with a build-in basic MQTT broker at: https://github.com/martin-ger/esp_wifi_repeater/tree/uMQTT_broker
You might give it a try. Currently it is build for MQTT V3.1.

Limitations: only QoS 0 and no saved sessions. Feedback is highly welcome.

maxboar, 2017-05-19:
Good day!
I have used option "save dhcp" to save 6 host MAC adresses and right after start all IP`s are ok. But after some time (when the dhcp lease set expires, I suppose) IP's are changed to not configured ones. Only first MAC address preserves IP.

Also I tried to forward 22 port of first host and connection starts, but no data is transmitted

martin-ger, 2017-05-19:
Will look into the  DHCP server...
What do you mean by "Also I tried to forward 22 port of first host and connection starts, but no data is transmitted". Portmapping doesn't work at all?
What is your config command for the mapping?

maxboar, 2017-05-19:
Config command was "portmap add TCP 22 192.168.7.2 22"
Http port mapping works fine, but ssh does not transmit data

aatta, 2017-09-10:
Hi, 
First of all big thank you for the great firmware, made my cheap ESP a capable range extender. I loaded the firmware and it started working right away. 

I know you have mentioned the use of port mapping to enable external world to talk to internal network, however I was wondering if we could have a way to use the uplink's dhcp server to get connected clients' ip addresses. In other words instead of running its own dhcp server, can the uplink's dhcp server be utilized in order to bring connected devices to the same network as the uplink network.

martin-ger, 2017-09-10:
This won't work: the ESP works as router, i.e. uplink and downlink network must be different IP networks. A DHCP server offers addresses from one network, that's why we need two and different addresses.
What you would like to have is a real repeater, i.e. layer 2 forwarding. Tried that, but as long, as I don't find an API to send a Layer 2 frame with the 'wrong' source address from an ESP's interface, this won't work.

chathurangawijetunge, 2017-09-11:
Hi nice job...

It is good to have DDNS update... to like
Noip.com
dynu.net

Thanks

aatta, 2017-09-11:
Thanks Martin for your detailed response to my query , I think I can get away with the portadd method just fine for now. 

BTW I have deployed ESP router just around the corner of my house and did some video streaming tests outside of my home, it worked quite well.

closed 2018-04-01 by martin-ger