martin-ger/esp_wifi_repeater #27: LED fast blink

yash0212, 2017-03-29:
I uploaded your binaries to my Nodemcu v1.0 (It has ESP-12E). after successful uploading, I restarted my device by removing and inserting the cable back. The LED on ESP blinked very fast.

I read in previously closed issues that the esp is rebooting but it had a solution for Wemos shield.

I am using Debian 8 Jessie and I tried your binary to install from Windows machine too, but the result was same.

martin-ger, 2017-03-29:
Device should be fine. Have tried to install any other third-party firmware before? Does this work? (Just in case there is a specific failure in your hw)

Does the serial port (115200 bps) give any reasonable output?

You might try to flash the files in the firmware_1.5.4 directory as an alternative. I sometimes had problems with sdk 2.0.1 but up to now only on ESP01-modules.

yash0212, 2017-03-30:
Other firmwares work perfectly like deauther of spacehuhn.

Serial port gives garbage values (weird icons).

I got an ESP12-E but the firmware 1.5.4 worked for me(I didn't try this before).
I uploaded the code using
`esptool.py --port /dev/ttyUSB0 write_flash -fs 32m 0x00000 0x00000.bin 0x40000 0x40000.bin`

What should be the password field if the ssid on which I want my repeater to connect is an open wifi named SRM-STUDENT.
I can't figure out the password(not ap_password) field in case of open wifi.
Please give the command

martin-ger, 2017-03-30:
Okay, I still do not understand the reason, why the firmware compiled with SDK 2.0.1 sometimes fails on some modules. Anyway...

Currently there is no option to set a static IP for the uplink side. Of course, it could be done in the sources.

martin-ger, 2017-04-04:
With version 1.4 you can set a static IP with:

- set ip ip-addr: sets a static IP address for the ESP in the uplink network
- set netmask netmask: sets a static netmask for the uplink network
- set gw gw-addr: sets a static gateway address in the uplink network

closed 2017-04-04 by martin-ger