martin-ger/esp_wifi_repeater #28: On set ap_ssid on ESP-01, the soft-AP network not change

alessandro308, 2017-04-08:
I've edit, via serial, the value of ssid and password. When I try to edit ap_ssid, the serial reply "changed" but the wifi network created not change so it remain unencrypted

martin-ger, 2017-04-09:
Works only after reboot. Older firmware versions also require a "set ap_open 0" in addition to the ap_password.

taner1, 2017-04-09:
reboot dont help for me. and i cant set any values with telnet or serial using latest precompiled binaries. show config command outputs  right but its still trying connect to "ssid". when reboot its going back to default too. also how can i configure open wifi that doesnt have password.

martin-ger, 2017-04-09:
Probably an issue of the power supply. Try to provide a stable 3.3V power, at least an additional capacity of several 100 µF parallel to Vdd . The power from the USB-to-serial converter is not enough for all modules.

taner1, 2017-04-09:
I just forgot to type save. It works like charm. set cpu to 160mhz and i got 8-9mbit on iperf tests.I dont even expect 1mbit from this 2$ module.  wow :+1:

closed 2017-04-10 by martin-ger