martin-ger/esp_wifi_repeater #29: Nat roughter

chathurangawijetunge, 2017-04-12:
Hi there....
The commands  "set ip" is not working

chathurangawijetunge, 2017-04-12:
I want a static ip to York nat roughter so I send

set ip 192.168.1.16
But I get invalid command 

Pls help...

martin-ger, 2017-04-12:
Did you "save" the setting? Only works after "reset".

martin-ger, 2017-04-12:
Do you use the lastest firmware? The command is rather new...

chathurangawijetunge, 2017-04-12:
Don't I get a confirmation when I send the command

chathurangawijetunge, 2017-04-12:
Where can I get the let us firmware

martin-ger, 2017-04-12:
For me it looks like this:

Trying 192.168.178.29...
Connected to 192.168.178.29.
Escape character is '^]'.
CMD>set ip 192.168.178.251
IP address set to 192.168.178.251
CMD>set netmask 255.255.255.0
IP netmask set to 255.255.255.0
CMD>show
STA: SSID:xxx PW:xxx
AP:  SSID:xxx PW:xxx IP:192.168.4.1/24
Static IP: 192.168.178.251 Netmask: 255.255.255.0 Gateway: 0.0.0.0
Clock speed: 160
MQTT: enabled
CMD>save
Config saved
CMD>reset

martin-ger, 2017-04-12:
Here in the firmware directory...

chathurangawijetunge, 2017-04-12:
Did not get the firmware directory....

chathurangawijetunge, 2017-04-12:
Pls post the link to firmware directory..

chathurangawijetunge, 2017-04-12:
When I try 
-->> set ip 192.168.1.16
I get 
<<--  Invalid command

chathurangawijetunge, 2017-04-12:
Pls post the new firmware directory

martin-ger, 2017-04-12:
https://github.com/martin-ger/esp_wifi_repeater/tree/master/firmware

chathurangawijetunge, 2017-04-12:
Thanks will try it tomorrow night let u know

chathurangawijetunge, 2017-04-13:
Hi friend Yorkshire new framework works very well... 😊😊😊

Do u have a method to upload to ddns server with the ip address 
I have done this with some lua script 
By sending API to server 

Like noip or dynddns etc

martin-ger, 2017-04-13:
Good to hear!

There is no DDNS-feature in the router today. Which address do you want to register? The uplink IP of the router? If you set it statically, anybody might register this know address.

chathurangawijetunge, 2017-04-13:
I want to update my home link roughter's ip with 
Dynu.com but my prolink roughter does not have Dynu.com as ddns servise provider 
so at the moment I use a esp8266 running a small lua script to do that job.

closed 2017-04-17 by martin-ger