martin-ger/esp_wifi_repeater #30: esp_uMQTT_broke: MQTT Invalid protocol

RibeiroSantos, 2017-04-20:
Hi,
When I try use with the MQTT Lens application is gives the message "MQTT: Invalid protocol".
Is just compile, upload, and create a client connection, yes?

martin-ger, 2017-04-20:
It requires MQTT v3.1.1, this message means that the client probably tried with v3.1.

RibeiroSantos, 2017-04-20:
Using MQTT Lens, we can select tcp: or ws.

tcp:  "MQTT: Wrong protocoll version" 
ws:   "MQTT: Invalid message    MQTT: Partial message received"

firmware compiled with 
//-#define PROTOCOL_NAMEv31  /*MQTT version 3.1 compatible with Mosquitto v0.15*/
#define PROTOCOL_NAMEv311     /*MQTT version 3.11

martin-ger, 2017-04-20:
Chrome MQTTLense seems to be v3.1 only. Used mqtt-spy for Java which does both.

Now uploaded a version that understands the v3.1 connect header. Seems to work for me with MQTTLense.

Switch is currently in the file mqtt/include/mqtt_msg.h

closed 2017-04-25 by martin-ger