martin-ger/esp_wifi_repeater #32: Unable to call xtensa-lx106-elf-readelf

metheis, 2017-04-23:
Hello there, I followed the instructions to build, and when I make the final repeater bin, its gives me the error:

Error calling xtensa-lx106-elf-readelf, do you have Xtensa toolchain in PATH?
Makefile:117: recipe for target 'firmware/0x00000.bin' failed
make: *** [firmware/0x00000.bin] Error 1

I made sure to change my BUILD_AREA to where the esp-open-sdk is

Here is the real path: /media/me/SD/Python/Wifi/repeater/esp-open-sdk/xtensa-lx106-elf

I attached a text version of the make file:

I verified that the xtensa-lx106-elf-readelf file exists, but I do not know why it cannot find it

[makefile.txt](https://github.com/martin-ger/esp_wifi_repeater/files/949379/makefile.txt)

martin-ger, 2017-04-23:
/media/me/SD/ is an SDCard? If so, is it formatted with VFAT or a Linux-FS? First guess: it doesn't have the permission bits?

Can execute the binary directly? Can you build the blinky example in the esp-open-sdk?

metheis, 2017-04-23:
Hello,

I cloned the repositories in the home folder, and now I receive a different error. When I make from the repeater source, I get:

CC driver/new_uart.c
make: /home/me/esp8266/esp-open-sdk/xtensa-lx106-elf/bin:/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: Command not found
Makefile:142: recipe for target 'build/driver/new_uart.o' failed
make: *** [build/driver/new_uart.o] Error 127

When I try and build the blinky:

CC driver/new_uart.c
make: /home/me/esp8266/esp-open-sdk/xtensa-lx106-elf/bin:/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: Command not found
Makefile:142: recipe for target 'build/driver/new_uart.o' failed
make: *** [build/driver/new_uart.o] Error 127

I have tried to re-clone the esp-open-sdk repository multiple times, but I still seem to get this error. Is there a problem with one of the open sdk files?

martin-ger, 2017-04-25:
Something wrong with the path: guest the gcc is at /home/me/esp8266/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc ? Please check the path again.

This is not a known problem of the esp-open-sdk.

RibeiroSantos, 2017-04-26:
Type "export" and see if you have the "xtensa-lx106-elf/bin" path in the PATH var.

metheis, 2017-04-29:
Hello, I was able to figure out the problem.

I needed to export the /home/me/esp8266/esp-open-sdk/xtensa-lx106-elf/bin PATH in the same terminal as I was MAKING. Sorry, I'm still new to Linux. I was able to flash the ESP8266. Now the only thing I'm having trouble with is setting up the repeater. I don't see the default wifi network and cannot connect via the serial port. how can I connect via STA during the initial setup?

Thank you for the help on the other procedure.

martin-ger, 2017-04-30:
What do you see on the serial output, when the router starts?

closed 2017-05-19 by martin-ger