martin-ger/esp_wifi_repeater #34: error when building with LWIP_DEBUG

st0ff3r, 2017-04-29:
tried to enable LWIP debug by editing esp-open-lwip/include/lwipopts.h
`#define LWIP_DEBUG			1
`

and running
`make -f Makefile.open clean
`

and recompile in esp-open-sdk
`make STANDALONE=y
`
when I try to build esp_wifi_repeater (or any other projects) I get this error:
```
/home/stoffer/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(dhcp.o): In function `dhcp_set_state':
dhcp.c:(.irom0.text+0x30): undefined reference to `ETS_ASSERT'
/home/stoffer/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(dhcp.o): In function `dhcp_option_short':
dhcp.c:(.irom0.text+0x62): undefined reference to `ETS_ASSERT'
/home/stoffer/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(dhcp.o): In function `dhcp_option':
dhcp.c:(.irom0.text+0xf3): undefined reference to `ETS_ASSERT'
/home/stoffer/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(dhcp.o): In function `dhcp_option_byte':
dhcp.c:(.irom0.text+0x17b): undefined reference to `ETS_ASSERT'
dhcp.c:(.irom0.text+0x1ce): undefined reference to `ETS_ASSERT'
/home/stoffer/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(dhcp.o):dhcp.c:(.irom0.text+0x1df): more undefined references to `ETS_ASSERT' follow
/home/stoffer/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(espconn_tcp.o): In function `espconn_server_recv':
espconn_tcp.c:(.irom0.text+0x6e4): undefined reference to `lwip_strerr'
espconn_tcp.c:(.irom0.text+0x813): undefined reference to `lwip_strerr'
/home/stoffer/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(espconn_tcp.o): In function `espconn_find_current_pcb':
espconn_tcp.c:(.irom0.text+0xd22): undefined reference to `ETS_ASSERT'
espconn_tcp.c:(.irom0.text+0xd3a): undefined reference to `ETS_ASSERT'
/home/stoffer/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(raw.o): In function `raw_sendto':
raw.c:(.irom0.text+0x12e): undefined reference to `ETS_ASSERT'
/home/stoffer/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(tcp_in.o): In function `tcp_receive':
tcp_in.c:(.irom0.text+0x4ce): undefined reference to `ETS_ASSERT'
tcp_in.c:(.irom0.text+0x556): undefined reference to `ETS_ASSERT'
/home/stoffer/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a(tcp_in.o):tcp_in.c:(.irom0.text+0x65e): more undefined references to `ETS_ASSERT' follow
collect2: error: ld returned 1 exit status
Makefile:249: recipe for target 'build/app.out' failed
make: *** [build/app.out] Error 1

```
any inputs on this?

martin-ger, 2017-04-29:
Never tried this option. Guess, these errors are not triggered by my changes. Obviosly, definitions of ETS_ASSERT and lwip_strerr are missing. No real idea...

martin-ger, 2017-05-19:
Thank you for fixing and extending this option!

closed 2017-05-19 by martin-ger