martin-ger/esp_wifi_repeater #35: New firmare with MQTT V 3.1.1

roboticboyer, 2017-05-06:
In the branch ESP uMQTT Broker could you upgrade the pre-compiled firmware so it supports MQTT protocol version v3.1.1?

The ESP MQTT client libraries now all support the version 3.1.1 and it isn't easy or it's impossible to use the V 3.1 

Thank you

martin-ger, 2017-05-07:
Latest commit is default MQTT v 3.1.1.

roboticboyer, 2017-05-13:
Thank You
The new firmware is working with the Arduino MQTT libraries using V 3.1.1

closed 2017-05-19 by martin-ger

roboticboyer, 2017-05-21:
I have written an article about my test with your library and the ESP8266 uMQTT Broker :
[http://wp.me/p4ZDu-3s](http://wp.me/p4ZDu-3s)

TridentTD, 2017-05-24:
#roboticboyer
Do you mind writting an article about  MQTT Client on uMQTT Broker ?
Thank you .

MQTT_CLIENT recompiled by 
[here](https://github.com/martin-ger/esp_wifi_repeater/blob/uMQTT_broker/user/user_config.h#L79)

martin-ger, 2017-05-24:
The current MQTT-Client functionality is NOT a MQTT bridge or something like that. Currently it only reports the internal status of the Router(Repeater) to an MQTT-broker. That is probably not what would you expect/need...

 

A bridge-functionality can be added separately: to publish local topics to an external broker is fairly easy. Subscribing external topics to republish them locally would require an additional config API. Not to mention mapping and renaming of topics (what e.g. a Moquitto Bridge can do).

 

Cheers,

Martin