martin-ger/esp_wifi_repeater #36: new firmware MQTT3.1.1 can't PUBLISH or SUBSCRIBE ?

TridentTD, 2017-05-09:
Previous firmware  I can connect PUBLISH or SUBSCRIBE to uMQTT on my esp8266.
but the new firmware  I can't.

I have tried by "mosquitto_pub" and "mosquitto_sub" ,
MQTT Broker can't connect as the following information.
---------------------------------------------------------
>mosquitto_pub -h 192.168.43.200 -p 1883 -t /home/room01 -m 13
Connection Refused: unacceptable protocol version.
Error: The connection was refused.

---------------------------------------------------------
>mosquitto_sub -h 192.168.43.200 -p 1883 -t # -v
Connection Refused: unacceptable protocol version.
Connection Refused: unacceptable protocol version.
Connection Refused: unacceptable protocol version.
Connection Refused: unacceptable protocol version.
....
....
---------------------------------------------------------

How to connect MQTT Broker on the new firmware?
Thank you.

BTW.
I set ESP8266 with new firmware by the following and reboot ESP8266 already
> set ip 192.168.43.200
> set gw 192.168.43.1
> set netmask 255.255.255.0

martin-ger, 2017-05-09:
New version should work with MQTT v 3.1.1 clients instead of v3.1. Sorry, I am out of the office for some days and have no chance to check into that. 

As this is the only change and if you experience problems, go back to the previous commit until I have a chance to dig into possible problems.

TridentTD, 2017-05-11:
Thank you.

I go back to the previous commit, V3.1, can use well.
But I found the other problem that 
when subscript to uMQTT by topic  wildcard  "#" ,
 the uMQTT v3.1 doesn't return the real topic from the publisher, 
it still returns topic "#" instead.

TridentTD, 2017-05-14:
#roboticboyer  Thank you.

I wish to control by Android App too. When you use with myMQTT or  MQTT Dash app on android, can you connected to mqtt 3.1.1 ?

roboticboyer, 2017-05-14:
I suggest to do some test with [mqtt-spy](https://github.com/eclipse/paho.mqtt-spy/wiki)
It's a GUI in Java

TridentTD, 2017-05-15:
#roboticboyer
Thank you.

martin-ger, 2017-05-18:
Thanks TridentTD, yes republishing the wildcard is wrong - fixed that in the latest verion.

Tested PUB/SUB with the mqtt-spy with MQTT v3.1.1 - no further obvious problems here.

closed 2017-05-19 by martin-ger