martin-ger/esp_wifi_repeater #37: Can the number of SUBSCRIBE more than 4? ( uMQTT 3.1 )

ShinDaiIchi, 2017-05-12:
It's very good work!

When I try uMQTT3.1 commit, (uMQTT3.1.1 can't run well for me)
I found that  SUBSCRIBE has limit only 4 subscribes.
Can more subscribes?

Thank you.

martin-ger, 2017-05-12:
The v3.1.1 seems to have a problem, will view into this next week, currentl I am on a business trip.

The number of Subscriber has three limitations: he number of WiFi clients on the internal AP is restricted to 8, the number of TCP connections has a limit in the lwip (maybe 4-5 as default) and the table of subscribed topics is limited (currently 50, user_main()).

You probably refere to the second limitation. Will try to compile a version with 8-10) TCP-clients next week.

svilex, 2017-05-14:
@martin-ger Is the **LmacRxBlk:1** error related to the number of TCP connections? I keep getting this error spammed in the serial monitor console. I googled it but i still can't figure out what it really means.

https://github.com/Links2004/arduinoWebSockets/issues/57
https://github.com/esp8266/Arduino/issues/2176
https://github.com/esp8266/Arduino/issues/928

TridentTD, 2017-05-15:
I found problem like this too.

I have tried on uMQTT3.1.1 broker , with MQTT-SPY (mqtt client 3.1.1/3.1)
I can connected to broker by only 5 differenct "mqtt-client id".

Thank you.

martin-ger, 2017-05-18:
**espconn_tcp_get_max_con()** gives you by default **5**.
Acording to the SDK this can be increased  **espconn_tcp_set_max_con()**. Tried with 10 - actually I managed 6 before memory shortage occured. So I guess 5 is more or less the maximum.

TridentTD, 2017-05-19:
I  tried by recompile with **espconn_tcp_get_max_con(10)** , however can only 5 clients

Feature request
- more than 10 clients
- bridge uMQTT to outside (on Internet) Broker.

Thank you again.

svilex, 2017-05-19:
@TridentTD If by client do you mean the device connected to the ap:
- [**espconn_tcp_get_max_con()**](https://github.com/martin-ger/esp-open-lwip/blob/sdk-1.5.0/lwip/app/espconn.c#L528) Just returns an uint8 and doesn't have any argument
- Maybe do you mean [**espconn_tcp_set_max_con()**](https://github.com/martin-ger/esp-open-lwip/blob/sdk-1.5.0/lwip/app/espconn.c#L543). Anyway, this function isn't related to the number of clients. What are you looking for i think is [this](https://github.com/martin-ger/esp_wifi_repeater/blob/master/user/user_config.h#L12)
- Keep in mind the esp has its own hardware limitations

@martin-ger I added **espconn_tcp_set_max_con(10);** [here](https://github.com/martin-ger/esp_wifi_repeater/blob/master/user/user_main.c#L1852) and seems like now i get not so many **LmacRxBlk:1** as before.

TridentTD, 2017-05-19:
Sorry , I miss type  .. I mean I compiled by  **espconn_tcp_set_max_con(10);**

svilex, 2017-05-19:
@TridentTD 
> bridge uMQTT to outside (on Internet) Broker.

For it to work outside I think you just need to port forward / port map

TridentTD, 2017-05-20:
@svilex 
Thank you.

martin-ger, 2017-06-05:
The max number of TCP connections can be increased with **espconn_tcp_set_max_con()** but in the repeater there is actually not much RAM left. I moved the project to https://github.com/martin-ger/esp_mqtt without the NAPT functionality (still the uMQTT broker can connect in both networks STA and SoftAP). I managed at least 8 connections there...

closed 2017-06-05 by martin-ger