martin-ger/esp_wifi_repeater #38: Can not access ESP-Repeater connected nodes!

foadem, 2017-05-14:
I'm extending my main access point using ESP-Repeater, I have a Domoticz server installed on main access point and I have an ESP node connected to ESP-Repeater. The access point IP is 192.168.1.1 and the ESP-Repeater IP is 192.168.4.1 and the ESP node IP is 192.168.4.122 However I can't open 192.168.4.122 url when I'm connected to main access point which is 192.168.1.1 I tried static routing but still I have no success.

TridentTD, 2017-05-14:
On the upper network can't connect to the lower network.
If you want to contact the server on 192.168.4.122 , you must set by PORTMAP (IP FORWORD)

----------------------------------------------------------------
portmap add [TCP|UDP] external_port internal_ip internal_port
portmap remove [TCP|UDP] external_port
----------------------------------------------------------------

Example
on the  UPPER Network  
Main AP : 192.168.1.1
You : 192.168.1.2    (connected to Main AP)
ESP Repeater : 192.168.1.3 (connecte to Main AP)

on the Lower network
ESP Repeater : 192.168.4.1
ESP-node :  192.168.4.122     ( connected to ESP Repeater   and have WebServer on port 80)

Set up PORTMAP on Esp-Repeater by serial-port (115200 baudrate) or "telnet 192.168.4.1 7777"
> portmap   add  TCP  8888   192.168.4.122  80 

When you (192.168.1.2 on the upper network)  want to connect  Webserver 
on ESP-Node (192.168.4.122 port 80  on the lower network).
You must call by ESP-repeater 192.168.1.3 with port 8888  instead.

http://192.168.1.3:8888

foadem, 2017-05-14:
Thank you so much, you are genius.

closed 2017-05-18 by martin-ger