martin-ger/esp_wifi_repeater #40: Unable to send pkts (ping) bigger than 1380 bits (or 1372Bytes?)

marek26340, 2017-05-17:
Hello! I absolutely love your project. It's helping me to get on the school wifi cuz it's crappy and always full, but this is automatically retrying to connect and it always connects! 👍
But i don't know why i can't send through pkts larger than 1372bits or bytes idk. I've got an NodeMCU v3 (ESP8266 ESP12E). I've set things up like this: CPU Speed: 160MHz; up/downstream: 0 (= unlimited).
Anyone knows why?

anudeepND, 2017-05-17:
Can you ping 1024 bites or less? It has something to do with MTU settings in the router itself

marek26340, 2017-05-17:
Ofcourse I can, but I can't ping >=1373 bytes. But I can ping 1372 bytes and less. I also tried multiple devices. On my laptop with windows, I can ping up to 1372 bytes, nothing more. Same on my android phone. But i can ping 1472 bytes with kali linux. (I wonder how it could do that? :D) Just to know, i'm trying to ping the esp itself, my home router and, for example, google. Everything cannot exceed 1372 bytes. If I will connect to my home router, everything is normal.

marek26340, 2017-05-17:
Even client to client ping is affected. This has to do with something like RAM, or buffer size etc.

anudeepND, 2017-05-17:
As far as I know, some routers have IPS (Intrusion Prevention System) when you ping it with more than 1024 bytes of ICMP packet, router thinks that it is being DDOSed and blocks it. In my case I can ping upto 1024 bytes, higher than this fails (because of IPS). However it is purely based on MTU value. And why do you need packet size more than 1024 bytes?

marek26340, 2017-05-17:
Well, i have a pretty cheap netis router at home. And i don't think that it has IPS. I just wanted to test, how much bandwidth it can handle before instability. And this limit also affects for ex. YT :D Really. This limitation extends the loading time to up to 1 minute until YT app realizes that it has to send more pkts at smaller size and etc. I mentioned above that i use it at school too, so this means, I've tested in more than one place/AP.

martin-ger, 2017-05-18:
I don't think, it is a problem of the esp_wifi_repeater. I can send bigger packets through it without any problems. Perhaps it is a problem with the MTU of your computer?

On Windows try: netsh interface ipv4 show subinterface
What does it tell you? Anything less than 1500 bytes?

marek26340, 2017-05-18:
It shows 1500 bytes. What else should it say? :D (executed that command while connected to the esp repeater). Oh and on my laptop, i can ping 1472 bytes no problem. Just a byte higher and it stops responding :D and on my phone, i can ping 1372 bytes max.

marek26340, 2017-05-18:
It doesn't respond even if i will ping it through the network that it is connected to. (so phone -> router ->esp) (Not even phone -> esp (-> router)) (1373 bytes and higher ofcourse)

martin-ger, 2017-05-18:
I am pretty sure that your client has an effective MTU size of 1400 (1372 + 28 Bytes ICMP header). The repeater definitely allows for 1500 bytes.

marek26340, 2017-05-20:
well idk what's the problem... maybe defective nodemcu? I really don't know :(

marek26340, 2017-06-03:
I think it's the ESP's problem. In serial console, i'm getting LmacRxBlk:1 .

martin-ger, 2017-06-05:
LmacRxBlk:1 means a memory allocation problem happened during the receive in the WiFi driver. An incoming packet is dropped. Happens during high load, but may happen in any IP network.

marek26340, 2017-08-13:
But that's the problem! It's dropping those packets that are bigger than 1372 bytes. This never happens to me, only when the packet's route consisted of phone, esp, router, roof antenna and etc. Well, i had my esp with a battery and a 5V USB booster, all packed together with regular tape. Looked very nicely. But some as****e in school took it from me and i have never seen it again. I think i was not fast enough with tracking the signal and that as****e unplugged the battery. 

I am still not that old, i'm 14yrs old, but i think i can do a lot of things with broken electronics/wires/etc. and a soldering iron. Recently i found old Genius computer speakers. Very old thing. The left speaker even had bad cables. So i took out that big transformer that was causing interference, glued on a 1300mAh 2s lipo and repaired that broken cable. Now i have portable speakers which last an eternity! Made them almost a month ago, playing music on maximum volume and the battery is still full. Wow.

I should be able to buy another NodeMCU within next month.

And i hope that with this new NodeMCU, i will finally be able to push packets bigger than 1372 bytes through it!

martin-ger, 2017-08-13:
What makes you sure, that the LmacRxBlk:1-Packets are those bigger than 1372 bytes? For me this happens from time to time in overload situations, but the stream recovers soon.

BTW, with LiPo batteries I use the plain ESP-12 without any converter (draws a lot of additional current), but just with a Si-diode as protection and at 3.3V. Works fine and is even cheaper.

martin-ger, 2017-08-13:
BTW. stealing an ESP is evil - guess they were after the batterie and had probably no idea, what the connected circuit does...

marek26340, 2017-10-13:
Well... LmacRxBlk:1 might be the problem, because the ESP is running out of memory while I'm trying to send packets bigger than 1372 bytes through it.

Yeah, I think so... I'm now at another school. That Gymnasium was sooo awful. I was just a student and they kept all the secrets (terrorizing my mother with lies! LIES!!) that I didn't know under the lid. argh

I still don't have a new NodeMCU. I'm currently giving money to my mom every week, because she had to buy me a new laptop for school and we all together don't have much money. I hope that I will have some money to spend on a new ESP until christmas..

marek26340, 2017-12-26:
Okay so I got a new NodeMCU v2. At first, it was getting into a boot loop, but I found that I have to flash something to address 0x3fc000 and it started working.

BUT the issue is still present! And I saw another open issue about LmacRxBlk:1 and no answers to be found anywhere.

I think that this is the limitation of the ESP8266 itself. Just take a look at it's RAM size. With some math, it turns out that it doesn't have enough RAM to store those packets that are larger than 1372 bytes and drops them.

marek26340, 2018-02-16:
I think that I just found something that looks like my problem... 

> https://github.com/ARMmbed/esp8266-driver/issues/18

and another.

> https://github.com/esp8266/Arduino/issues/1009

I think that I should close this issue because this is not a problem in the code, but it could be something in the espressif libraries/SDK/whatever, or, most probably, it's a limitation in the ESP itself (most likely RAM/buffer size).

devyte, 2018-02-16:
There are known issues with udp in lwip1.4. It is the main reason we (esp8266/Arduino repo) migrated to lwip2.

mtlive, 2021-02-14:
> There are known issues with udp in lwip1.4. It is the main reason we (esp8266/Arduino repo) migrated to lwip2.
There is a similar project for ESP32 which uses the newer SDK (idf) and also newer lwip.
I tried it on ESP8266 (using the new FreeRTOS SDK which supports idf style) and could successfully compile and run it on a NodeMCU, but unfortunately there is no internet connection, even though the ESP8266 connects to the router.

More details:
jonask1337/esp-idf-nat-example#18