martin-ger/esp_wifi_repeater #41: OTA update

brandobur, 2017-05-20:
Perfect job! Absolutely the best esp8266 WiFi NAT Router!

Is it possible to update it by OTA update? It will be great if yes, because in my case for expamle, I will assembly my box with esp8266 to the wall in garage and standard way is very complicated.

I can image for expamle this solution [HTTP Server](https://esp8266.github.io/Arduino/versions/2.0.0/doc/ota_updates/ota_updates.html#http-server).

Thank you very much.

martin-ger, 2017-05-21:
Guess, it is the only NAT router available ;-)

Havn't looked into the OTA-features so far. The description referes to the Arduino environment. If anybody has experiences or even better code on how to do it with the NONOS SDK I will try to integrate it.

brandobur, 2017-05-21:
I am totally busy next 2 months, unfortunately, but if i will have some time, I will try it.

martin-ger, 2018-09-12:
OTA is supported now!

closed 2018-09-12 by martin-ger