martin-ger/esp_wifi_repeater #42: How to flash into ESP8266

toeybaa, 2017-05-24:
Hi guys, I am very new to use this kind of thing here. Sorry for that, as I have read all of your md file. I found it is interesting for me to use your project. But what I don't understand is that how to upload your project into ESP8266 model. I use Windows OS. Do I need to compile all the project?. What I have done is download those 2 bin files 0x00000 and 0x10000 and input them into Download tool V.3.44 and set parameters according to your sample. But the error log said 
====================
EFUSE NORMAL MODE
====================
EFUSE CHECK PASS...
48bit mac
debug:
-----------------------
 5c cf 7f f0 2c ca
CUSTOM_ID: 06 40 00 00 00 09 40 00
CUSTOM_ID: 0640000000094000
-----------------------
crc_efuse_4bit: 0
crc_calc_4bit: 14
48bit mac
MAC AP : 5E-CF-7F-F0-2C-CA
MAC STA: 5C-CF-7F-F0-2C-CA
get mac res:  True
('tttest uuuuuuuuuuart : uart reg: ', 458)
(' baudrate: ', 115200)
get crystal: 26380800
get flash id : 0xff1640e0
 manufacturer_id: 0xe0

 device_id: 0x4016

vendor: 224
mode: 64
size: 22
DEBUG!!!!
SET FLASH PARAMS
filename:  D:\Users\Peth\Desktop\FLASH_DOWNLOAD_TOOLS_V3.4.4\bin_tmp\downloadPanel1\0x00000.bin_rep
offset :  0
filename:  D:\Users\Peth\Desktop\FLASH_DOWNLOAD_TOOLS_V3.4.4\bin_tmp\downloadPanel1\0x10000.bin
offset :  65536
Erasing flash in old style...
*********************
pic path: ./RESOURCE/DOWNLOAD_S.bmp
***********************
Traceback (most recent call last):
  File "download_panel_info.pyo", line 539, in update_pic
AttributeError: ESPDOWNLOAD instance has no attribute 'ESP_MAC_BT'
Took 0.47s to erase flash block
Erasing flash in old style...
Took 1.92s to erase flash block
TEST !!!!
FAST BAUD: 230400
fACTORY REBOOT MODE:  False
Running Cesanta flasher (speed 230400)...
params:  
MEMORY START
ELSE.... 26
Invalid head of packet ('')
Exception in thread Thread-2:
Traceback (most recent call last):
  File "threading.pyo", line 801, in __bootstrap_inner
  File "download_panel_info.pyo", line 785, in run
  File "esptool_api.pyo", line 763, in flash_download_thread
  File "esptool_api.pyo", line 1761, in __init__
  File "esptool_api.pyo", line 1646, in read_packet
Exception: Invalid head of packet ('')

Could you please figure me out. Thanks

jadolg, 2017-07-07:
hey there. I just flashed my ESP8266 and it works like charm. Install esptool (it uses python)
`pip install esptool`
and then:
`esptool.py --port /dev/ttyUSB0 write_flash 0x00000 firmware_sdk_1.5.4/0x00000.bin 0x40000 firmware_sdk_1.5.4/0x40000.bin`
I had to flash firmware_sdk_1.5.4 but it works perfectly. (I did this on a Linux machine (Ubuntu 16.04))

GoranVidakovic, 2017-11-19:
I use ESP8266 download tool and send two binary files to ESP. After restart they blinking but can not find network to connect to. Did I need to sent to ESP8266 anything else of files in folder?

james-bell, 2018-01-13:
I have this exact problem "Invalid head of packet ('')" with the NodeMCU but I'm on Windows, is there a way to flash this with Windows?

GoranVidakovic, 2018-01-13:
Actually I solve the problem.
I'm windows user too. Download and install ESPFlashDownloadTool_v3.6.2 and
install on Windows.When you start application press bottom ESP8266
DownloadTool. If you use NodeMCU you need to download to 00000.BIN and
40000.BIN files and put him on same address with Downlaod tool. Files are
in file firmware_sdk_1.5.4
After uploading BIN files follow instructions from web to setup repeater.
Regards
Goran

james-bell, 2018-02-02:
So I downloaded ESPFlashDownloadTool_v3.6.3 and was able to flash the 1.5.4 firmware to 0x0 and 0x4 but nothing happened.  I then flashed the latest 0x0 and 0x1 firmware from this project.  Now my NodeMCU's LED flashes constantly but I don't see any WiFi access point per the instructions.  I'm not even sure the flashing tool worked properly, all I get at the end of a flash is "48 bit mac".  Then I tried the "erase" option on the tool and flashed again, no difference.

martin-ger, 2018-02-03:
Did you try DIO mode of the flasher? If so: did you try that:

Sometimes, especially on old ESP-01s, there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/espressif/ESP8266_NONOS_SDK/raw/master/bin/esp_init_data_default_v08.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

james-bell, 2018-02-04:
Mine is a v.2 NodeMCU.  I admit I'm a complete novice at this why does the manufacturer tolerate crappy flashing tools with bad feedback and mysterious settings?  I'd think they would make a rock solid tool to help us more.

GoranVidakovic, 2018-02-04:
Connect to ESP with serial monitor like Arduio IDE, chose right baud rate
and send instructions for changing and activating wifi. Then connect thru
wireless connection. Instructions are on the web page. By the way
everything works fine with ESP8266 any type.

Regards
Goran

Oxyandy, 2018-03-06:
[image]
Downloaded the latest files today & now struggling I have read different things older things & ahh
Ok I have the files above... I would love to flash a recent Sonoff Basic it takes a 1M flash 
say I used esptool.py - what command line should I use ? 
I have been using 1M firmwares and everything has gone well
but these are in pieces 
here's what works - change anything ?
esptool.py.exe --port COM5 write_flash -fs 1MB -fm dout 0x0My_Firmware.bin
Thanks, excited to try this out on Sonoff & happy to give feedback
Cheers

martin-ger, 2018-03-06:
Use:
For each firmware version:
esptool.py --port COM5 write_flash -fs 4MB -ff 80m -fm dio 0x00000 
0x00000.bin 0x10000 0x10000.bin

And once for the Sonoff:
esptool.py --port COM5 write_flash -fs 1MB -ff 80m -fm dio 0xfc000 
esp_init_data_default_v08_vdd33.bin

Oxyandy, 2018-03-06:
wow so fast ! Thanks Martin
MUCH appreciated

Oxyandy, 2018-03-06:
STAGE 1 - Looks OK
`esptool.py --port COM5 write_flash -fs 4MB -ff 80m -fm dio 0x00000 router\0x00000.bin 0x10000 router\0x10000.b
in
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 48876 bytes to 12029...
Wrote 48876 bytes (12029 compressed) at 0x00000000 in 1.1 seconds (effective 365.1 kbit/s)...
Hash of data verified.
Compressed 48876 bytes to 12028...
Wrote 48876 bytes (12028 compressed) at 0x00010000 in 1.1 seconds (effective 364.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...`
STAGE 1 - Looks BAD :(
`esptool.py --port COM5 write_flash -fs 1MB -ff 80m -fm dio 0xfc000 router\esp_init_data_default_v08_vdd33.bin
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...

A fatal error occurred: File router\esp_init_data_default_v08_vdd33.bin (length 49203) at offset 1032192 will not fit in 1048576 bytes of flash. Use --flash-size argume
nt, or change flashing address.`
Thanks again ;)

martin-ger, 2018-03-06:
Okay, Sonoff has only 512k - use address 0x7c000

Oxyandy, 2018-03-06:
OK, that looks happier..
```
esptool.py --port COM5 write_flash -fs 1MB -ff 80m -fm dio 0x7c000 router\esp_init_data_default_v08_vdd33.bin
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 49204 bytes to 12040...
Wrote 49204 bytes (12040 compressed) at 0x0007c000 in 1.1 seconds (effective 367.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
```
Should be LED ? Nudda
192.168.4.1 took too long to respond.
Serial output at 150000 baud
r␘␂␀l��r�␀�#␂�n�␄��␌␘�␌␜��␜p�|���␀�8␂��ǒ��l␜p␌␘␌�nn�␂�;�nĒ��␌␛�␌b�#l`␛$`␛$��r���w␂␒n��bp␀�
Next, (other that me saying thanks for your time & this should help others)
The firmwares I flash & backup are 1024 Kb
Cheers

martin-ger, 2018-03-06:
On the Sonoff: No LED by default
Use 115200 (!) on serial: should look more readable: Something like version 
info.

AP "MyAP" should appear on WiFi...

Oxyandy, 2018-03-06:
Yes sorry I used the right baud before, just typed it wrong
I just tried changing DIO to DOUT & redoing both..
Should I use the erase command too?
```
--- Miniterm on COM5  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
␀r␘␂␀l��r�␀�#␂�n�␄��␌␘�␌␜��␜p�<���␀�8␂��ǒ��l␜p␌␘␌�nn�␂�;�nĒ��␌␛�␌b�#l`␛$`␛$��r��Iw␂␒n��bp␀
```
Thanks

martin-ger, 2018-03-06:
Think, erase ist not required. Can you flash any other binary? E.g. Tasmota?

Oxyandy, 2018-03-06:
Yes, I flash ESPeasy, ESPurina & Tasmota etc no problem 
& make use of configured back-ups
Thanks

Oxyandy, 2018-03-06:
```
esptool.py.exe --port COM5 erase_flash
esptool.py v2.2.1
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 2.8s
Hard resetting...
```
Starting fresh

martin-ger, 2018-03-06:
You should ist the same parameters...

Oxyandy, 2018-03-06:
Just tried 
`esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x00000 router\0x00000.bin 0x10000 router\0x10000.bin`
Powered Sonoff off, then On again into flash and 
sptool.py --port COM5 write_flash -fs 1MB -fm dout 0x7c000 router\esp_init_data_default_v08_vdd33.bin
Now serial only says #␀#

Oxyandy, 2018-03-06:
If parameters are the same I should be able to do one line ?
`esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x00000 router\0x00000.bin 0x10000 router\0x10000.bin  0x7c000  router\esp_init_data_default_v08_vdd33.bin`

martin-ger, 2018-03-06:
Doesn't look good. Can you check the SHA-1 checksums of the binary files?

It definitly runs on a Sonoff Basic...

martin-ger, 2018-03-06:
Yes, one line is fine.

Oxyandy, 2018-03-06:
ah you are right no need for SHA1 files.. I just opened in a text editor and oops html only how could I not notice the size too... sorry

Oxyandy, 2018-03-06:
`esptool.py --port COM5 write_flash -fs 1MB -fm dout 0x00000 router\0x00000.bin 0x10000 router\0x10000.bin 0xfc000 router\esp_init_data_default_v08_vdd33.bin`
Now the serial out is saying strange stuff
```
WiFi Repeater V1.7.2 starting

No config found, saving default in flash
Loading default configuration
Saving configuration
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(5c:cf:7f:72:98:03) + softAP(5e:cf:7f:72:98:03)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
```
THANKS so much Martin, best wishes for you - a nice single to flash

Oxyandy, 2018-03-06:
Am I allowed to find bugs ?
Should I open a new bug report ?
Plus is there a manual somewhere, I have been making use of help cmd in serial
obviously there is much more to this..
Should the access point have the same number as my home router ?
I try change it, but it stays same (anyhow basic users guide would be cool)
Thanks for all the time you put into this project Martin, legend

martin-ger, 2018-03-06:
Please report any suspected bug in a new issue. You are welcome to improve 
the project.

Up to now the Readme.md ist the documentation.

The networks should bei different:
set network x.x.x.x
save
reset

Oxyandy, 2018-03-06:
Thanks Martin, another day here and I am feeling a little bit foolish, 
the Readme.md is going to be most useful, I will read it over many times I am sure.. 
You are bound to get some interesting feedback from me because you have encouraged me to 
"improve the project." - This makes me feel like my words/feedback will not be wasted, so will go into detail as much as possible.. Saving a local copy to compile at home.
I really appreciate the time you have given responding to me & sorry I messed some of it up, I usually do not right click links and "Save Target As".. (html bins) - wont do that again, cheers

Oxyandy, 2018-03-06:
ESP WiFi NAT Router Config
AP Settings
Attempting to change the SSID: results in
The new settings have been sent to the device...
After a little bit page reloads, but AP SSID does not change ?
It is currently the same SSID as STA Settings
Is it intentional that they are the same ?
I am attempting to get my far away outdoor devices connected as Primary Network, with my home router set in these far away devices as secondary..
Anyway it might be mentioned in the Wiki, I will go search that for " SSID " and read what I find, otherwise is a bug..
I tried doing this while connected to my serial cable, but admit I did not send the save command..
Off to read & soak it in

Oxyandy, 2018-03-07:
[image]

# Followed by reset - then show cmd again

[image]

# Why AP SSID will not stick ? waaaa

martin-ger, 2018-03-07:
You have 'automesh' ob (1). With automesh STA and AP seid are always equal.

Do
set automesh 0
If you want a individual ap_ssid

Oxyandy, 2018-03-07:
I never changed it Martin, it must be set default 'on' in the source, anyway that is neither here nor there if the read me had a little extra info, my head would hurt less.. Kept thinking I was doing something wrong with the way I was setting things
As newbie I have the prospective of one, well obviously ;)
And hope you do not mind my feedback ramblings of where I am getting stuck, it might give you some insight, so maybe I could so some 'Read me' edits ? 
Some extra words & some rewording maybe needed in a few places
example something like this (pic below) would have helped & doesn't this text also need the step 
`set ap_open 0`
too ?
[image]

martin-ger, 2018-03-07:
Top check it: please try 'reset factory' .
This resets to default values (same function at first boot) , should switch 
automesh off.

Oxyandy, 2018-03-07:
Hi again, a few things..
The most important one - I am running ESPeasy Firmware on 2 devices I want to extend the range on through another Sonoff running your firmware - not been able to once yet..
Now after much thought, maybe it is because I have them set as static IPs 192.168.0.200 & 192.168.0.250
They connect but I can never view the web configuration page, yes I know I can remove the static IP assignments and try again, but I use those for quick, hassle free direct access - if I ever needed to access them, which at the moment is frequent enough..
I see 3 connected devices (1 is my main PC & I can get to the net through that)
see image
[image]
I assume the 2 without MAC addresses are my devices...
Options ? please, thanks again

martin-ger, 2018-03-07:
Yes, devices without DHCP mapping so not appear in the list - but they 
should work anyway.

But even If you use static addresses, these must belong to the repeaters AP 
subnet, default 192.168.4.x. You might change that subnet, but it should 
bei different from the upstream IP Network.

Oxyandy, 2018-03-07:
Ok strange, while 2 were connected - each with a static IP assignment (in the device itself), 
I could never type the IP into the address bar and connect, I removed the assignment and just 'one' and now they both show in the browser :O one as it's proper http://192.168.0.250 
the other assigned DHCP http://192.168.0.2 - both with RSSI figures under -70 - so , much better
About to undo the change as see if it reverts to being unable to access either device's webconfig page..

Oxyandy, 2018-03-07:
What I should have added before
[image]
See that check box, ignorance by me may have checked it the first time round, so when I said before this might be enabled by default in the source, might not be true at all - learning..
But I am seeing in read me 
`set phy_mode [1|2|3]: sets the PHY_MODE of the WiFi (1=b, 2=g, 3=n(default))`
and finally found repeater is 802.11g, my computer doesn't care, but seems my router hates it, so changing that to n now like everything else here.
Now to assign both static again and see if it reverts, but `set phy_mode 3`first

martin-ger, 2018-03-07:
SationAP mode only uses g-mode, even when n ist configured.

Oxyandy, 2018-03-07:
set phy_mode 3
WOW, huge difference pages from the internet loaded slowly now BAM
complex little sucker to fine tune

Oxyandy, 2018-03-08:
Hello Martin, I must be doing something wrong
As it is now, I have my home network & repeater as different SSIDs
I returned my devices to static and they both connect to the repeater as preferred network, if I join the repeater with my PC, as static I get  both devices web pages to load quickly
That part works good.. If I try to load an external webpage nothing loads. (no internet access)
repeater is showing as connected to my home router - has host name ESP_xxxx & IP
If I leave my devices connected to repeater and shift my PCs wireless connection back to home network & type the devices IPs - nothing opens
Try repeaters IP again doesnt load

Here is what I want to do, I want to just use the repeater as a repeater to boost signals on them both so they can be accessed on main PC at a better strength.. Type in their IP and they load..
Do I have the wrong mode? Suggestions please

Oxyandy, 2018-03-08:
Automesh mode
http://192.168.0.102 Repeaters page loads, devices show as connected but their pages will not load on main PC connected on same router as repeater.
[image]

martin-ger, 2018-03-08:
Let me try to explain: let's call the uplink side to the internet (STA of the ESP) "outside" and the other side, tha AP side of the ESP where the clients are "inside". Those two sides have different IP subnets and addresses.
Nodes from the inside can connect to the internet as their addresses are translated via NAT to the outside address of the ESP. Their requests look to the outside node as if they would come from the ESP itself. This translation is learned dynamically from the requests and it is stored temporarily in the NAT tables of the ESP. Replies are translated again in the reverse direction to the inside addresses.

But you cannot directly access inside nodes from the outside, as there is not translation known. What you have to do is: define a static translation from the outside addr/port of the ESP to the inside addr/port of the nodes. I.g.: your outside addr of the ESP is 192.168.0.2, the IP address of the internal node is 10.24.1.2, you want to do a web-request to the inside node from your PC outside. Define a mapping:
- portmap TCP 80 10.24.1.2 80
and access the outside addr of the ESP: http://192.168.0.2:80. You will be redirected to the inside node port 80.

If you have a second inside node, let's say 10.24.1.3, use a different outside port and do:
- portmap TCP 81 10.24.1.3 80
and access the outside addr of the ESP: http://192.168.0.2:81. You will be redirected to the second inside node port 80.

If you inside addresses are static this is fairly easy, if you use DHCP for dynamic addresses here set it all up once and use:
- save dhcp
to save the current assignment of the inide addresses.

Hope this explains. Transparent bridging on ISO layer 2 would avoid that (what a real WiFi REPEATER does) but as far as I know today, this cannot be done with the ESP drivers as they are...

Oxyandy, 2018-03-13:
Thanks again Martin, excellent explanation, I had to get another router & configure it to an access point, the sonoffs I am using control a garden so I had to resolve connectivity first, in a few days I will return to see if I can get you code running again.. 
Just a bit behind on other chores - 
Cheers

closed 2018-04-01 by martin-ger

Rizwan-Hasan, 2020-11-06:
> STAGE 1 - Looks OK
> 
> ```
> esptool.py --port COM5 write_flash -fs 4MB -ff 80m -fm dio 0x00000 router\0x00000.bin 0x10000 router\0x10000.b
> in
> esptool.py v2.2.1
> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Uploading stub...
> Running stub...
> Stub running...
> Configuring flash size...
> Compressed 48876 bytes to 12029...
> Wrote 48876 bytes (12029 compressed) at 0x00000000 in 1.1 seconds (effective 365.1 kbit/s)...
> Hash of data verified.
> Compressed 48876 bytes to 12028...
> Wrote 48876 bytes (12028 compressed) at 0x00010000 in 1.1 seconds (effective 364.7 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting...
> ```
> 
> STAGE 2 - Looks BAD :(
> 
> ```
> esptool.py --port COM5 write_flash -fs 1MB -ff 80m -fm dio 0xfc000 router\esp_init_data_default_v08_vdd33.bin
> esptool.py v2.2.1
> Connecting....
> Detecting chip type... ESP8266
> Chip is ESP8266EX
> Uploading stub...
> Running stub...
> Stub running...
> Configuring flash size...
> 
> A fatal error occurred: File router\esp_init_data_default_v08_vdd33.bin (length 49203) at offset 1032192 will not fit in 1048576 bytes of flash. Use --flash-size argume
> nt, or change flashing address.
> ```
> 
> Thanks again ;)

Stage 1 Works

martin-ger, 2020-11-06:
What kind of ESP8266 do you have? Why "-fs 1MB" in the scond command?

Rizwan-Hasan, 2020-11-06:
### **This one worked for me**
**esptool.py--port_ /dev/ttyUSB0 write_flash -fs 4MB -ff 80m -fm dio 0x00000 router\0x00000.bin 0x10000 router\0x10000.bin**

[image]
[image]

martin-ger, 2020-11-06:
Normal ESP12, should use "-fs 4MB"

Rizwan-Hasan, 2020-11-06:
I actually quoted a comment of Oxyandy from there.