martin-ger/esp_wifi_repeater #43: web login

janpsegers, 2017-05-26:
Is it possible to web login on the existing network ?

martin-ger, 2017-05-26:
Don't really understand what you are asking. The repeater itself has no web interface so far. Web traffic is forwarded without any problem. To access a server in the NATed subnet, you will need a port forward, e.g.:
portmap TCP 80 192.168.4.2 80

janpsegers, 2017-05-26:
When i am connnected to a hotspot i am used to see a login page when i 
first acces the internet.  This login is not shown when i range extend the 
hotspot with your router although i can see the router is connected.

Verzonden met AquaMail voor Android
http://www.aqua-mail.com

martin-ger, 2017-05-26:
The login-page of an open hotspot is usually implemented by a captive portal, i.e. a DNS redirect. When the esp_wifi_repeater connects, it just does the WiFi-connection without any Web- or DNS-request. But when you now connect to the esp_wifi_repeater with a browser (smartphone or laptop), it will do an DNS-request and it will show up with the login page. You only have to login once, as this registers the MAC and the IP of the esp_wifi_repeater, which will be reused ba all other devices that connect through it later. Hope this explains...

janpsegers, 2017-05-26:
I expected the mechanisme you discribe. When i open the browser it tries 
google and should be redirected but in fact it stops and times out.  The 
redirection should go to a network ip on port 8880 .
I will first try an an other hotspot to see if i can get a connection
 Thanks so far

Verzonden met AquaMail voor Android
http://www.aqua-mail.com

janpsegers, 2017-05-29:
The problem is partly solved.  Its probably caused by low wifi signal. 
Closer to the wifi ap i could connect to the hotspot so we can close the 
topic. What should the power value show in the show stats list.

Verzonden met AquaMail voor Android
http://www.aqua-mail.com

closed 2017-08-16 by martin-ger