martin-ger/esp_wifi_repeater #44: uMQTT 192.168.4.1

roboticboyer, 2017-05-28:
Hi
I'm making new tests, but I have fond an issue
The ESP MQTT client is able to connect to the ESP with uMQTT only using the Station IP not the AP 192.168.4.1
There is the TCP connection, but the MQTT connection doesn't start 

I have tried with 2 different libraries for ESP MQTT client
I'm using this test [code](https://gist.github.com/roboticboyer/b47236a097a61984db24a968e993af42#file-mqtt_test-ino)

martin-ger, 2017-05-28:
Tried to connect from both sides simultaneously - from AP and STA with the IP of either side (). Works just fine. Maybe you experience problems with MQTT version?

roboticboyer, 2017-05-29:
Hi Martin
I'm using your last version of the firmware (MQTT 3.1.1 compliant) with the below ESP8266 Mqtt libraries:
[https://github.com/256dpi/arduino-mqtt](https://github.com/256dpi/arduino-mqtt)
[https://github.com/knolleary/pubsubclient](https://github.com/knolleary/pubsubclient)

When I connect using the STA IP all works fine, while when I use the AP IP the 2 libraries return that the MQTT connection has not established.
My concerns are:
- using STA IP all works, so there isn't any problem with my ino code
- also It's strange that the problem exist with 2 different libraries (one is not derived from the other one)

How you have made your test? I could tray to replicate it.

How can I download the previous version of your firmware compliant only with MQTT 3.1?

Thank You

roboticboyer, 2017-05-29:
I have done a further test with python using my linux PC: [MQTT_Publish_Test_00.py](https://gist.github.com/roboticboyer/61e4e09526fd89287d63122d3a23b756)
It works with AP and STA using paho.mqtt.client

The issue is something related to the ESP8266 implementation of the MQTT libraries
Do you have experience with them?

roboticboyer, 2017-05-30:
I have simplified the [test ino file](https://gist.github.com/roboticboyer/b47236a097a61984db24a968e993af42#file-mqtt_test-ino)
If ESP8266 connects the STA IP there is the MQTT connection.
If ESP8266 connect the AP IP(192.168.4.1)  now there is the MQTT connection but I have added 
[the row](https://gist.github.com/roboticboyer/b47236a097a61984db24a968e993af42#file-mqtt_test-ino-L36)
`https://gist.github.com/roboticboyer/b47236a097a61984db24a968e993af42#file-mqtt_test-ino-L36`

This is quite strange but works!

martin-ger, 2017-06-01:
As the WiFi mode seems to be the crucial point, I would guess that it is really a lower level problem, not one of the MQTT protocol.

Just checked in a version that accepts MQTT v3.1 and v3.1.1 clients at the same time.

martin-ger, 2017-06-02:
Migrated the uMQTT broker into a new project without the NAT feature: https://github.com/martin-ger/esp_mqtt 
This saves RAM and allows for more TCP connections. I tested with 8 yesterday. Also it simplifies the compile process as it doesn't require the modified lwip library any more, the standard SDK should be sufficient

roboticboyer, 2017-06-02:
I have updated my [blog](http://wp.me/p4ZDu-3s) with your new Git version

martin-ger, 2017-06-03:
Hi,

 

thank you for your interest and your help.

 

I am currently working on an extension of the broker towards an MQTT-bridge, i.e. that can work also as an MQTT client to another broker and forward topic messages. This could help to collect the messages from a local MQTT domain (private WiFi network) and securely pass them to a remote broker in the internet (possible even via TLS).

 

Cheers,

Martin

roboticboyer, 2017-06-04:
Hi Martin
Good Job! The MQTT-bridge it's a good idea!
Another application is the 1st uMQTT broker sends the messages to another uMQTT broker to extend the ESPs network

hjahend, 2017-11-27:
Hello  Martin, 
I am very interested in your MQTT Broker/Bridge on the ESP8266 ); https://github.com/martin-ger/esp_mqtt  

For about 15 years I am using Homevision (http://www.csi3.com/homevis2.htm  ) to control my home lighting  via X10 transmission and several other tasks. Now I am transferring my hallogene lights to LED (they consume less power), also using Sonoff Wifi units and Tasmota software (see github) for Wifi control. That works already quite well, but it also needs a PC, via RS232 attached to the Homevision unit  24/7,  in order to transmit the Homevision commands via HTTP and/or MQTT to Sonof Wifi units.  
I have several Sonoff units working and they all need to be controlled via Wifi reliable, simple and if possible: no PC 24/7. I use a local mqtt broker on a PC now.

Separate is the e-mail correspondence with HomevisionXL expert Ron Boston (USA), who made an MQTT plugin between Homevision and a permanently attached PC (see: attached file  and (http): https://github.com/rebel7580/ESP8266-Wemo-and-HomeVisionXL-Plug-in  and (tcp): https://github.com/rebel7580/ESP8266-Alexa_Skills_Thermostat ). 
Ron thinks that your solution could be implemented, so eliminating an attached PC with mqtt on it. 
But I am not a coding guy and do need assistance/guidance from an expert like you.
Can you help me out in this matter?
If it all works well and is simple to implement a lot of other Homevison users may also be interested. 
The main execution service that I need from the ESP 8266 mqtt broker is  ‘the Power On and Off’ instructions for the Sonoff units and possibly: its status feedback (is a Homevision flag set/clear). 
Extra features (like temperature values transfer, etc) are not needed via wifi in my situation. 
Waiting for your reply and kind regards,
Harry Hendriks
The Netherlands
[2017 HomevisonXLplusMQTT.docx](https://github.com/martin-ger/esp_wifi_repeater/files/1505964/2017.HomevisonXLplusMQTT.docx)

martin-ger, 2017-11-27:
Hello Harry,
I would love to see the esp_uMQTT_broker working in your environment and if can, I would like to support this. 

What I don't understand so far is the exact setup of your system: The Homevision box has a serial interface and talks to a PC? Who issues the MQTT requests? The PC as result of the serial input? The Homevision itself? Who runs the plugin?

The esp_uMQTT_broker as is can send text to the serial output (but it will probably send some other output as well) but it currently cannot read input from there (other than CLI commands). Probably we'll need a special program that can do the job...

hjahend, 2017-11-28:
Hello Martin
See my comments in your email below.

Hello Harry,
I would love to see the esp_uMQTT_broker working in your environment and if can, I would like to support this. Thank you very much!

What I don't understand so far is the exact setup of your system: The Homevision box has a serial interface and talks to a PC? Who issues the MQTT requests? The PC as result of the serial input? The Homevision itself? Who runs the plugin?
See http://www.csi3.com/homevis2.htm .  When configured properly Homevison (HV) does not need an attached PC because it has an eprom in it and can operate self contained.
But you do need a PC to alter the instruction software or other functions, change values, monitor the status, send and receive commands, upload a new program in the HV eprom, etc. etc.
The communication between HV and PC is via RS232. This Two-way serial interface provides full control and status reporting via the PC. You can also Run (extra) programs to add voice recognition, voice response, and many other features.
There is also a second RS232 port available for serial command transfer.
A HV serial port connector uses RS-232 voltages levels.  So we will need an RS-232 transceiver to convert to 3.3V TTL levels of an ESP 8266.  HV itself uses a MAX232CPE to convert internally to 5V TTL levels.  I still have to look for an appropriate and simple  3.3V RS232 transceiver when connecting HV to an ESP 8266 (any suggestions?).

The esp_uMQTT_broker as is can send text to the serial output (but it will probably send some other output as well) but it currently cannot read input from there (other than CLI commands). Probably we'll need a special program that can do the job...
Ron Boston (HV expert who wrote several HV plugins, including mqtt message control) suggested that the scripting feature and its "print/println" commands in your code could possibly be used to send the appropriate commands/results to a serial connected HV.
If his assumption is correct, this approach may not require ANY modifications at all to your code; only add in the appropriate scripts (but I do not know how and why..)!
See for more details the mqtt examples in the previously attached document.

If this all is not clear enough yet we may have to consult Ron Boston to help us out.

Ron Boston told me that he is quite interested in this project. This also because it could be helpful for many other HV users.  To his regret Ron does not have the time now to get deeply involved in writing code but he is able to consult with both of us on any HV issues/questions in this project.
I added Ron to the cc list of this email to keep him posted.

Kind greetings

Harry Hendriks

hjahend, 2017-11-28:
Harry/Martin -

There is a (big) hole in my thinking about this so far -- I've been concentrating on how the esp_MQTT sw can pass device state/status info TO HomeVision, neglecting the obviously important control of the devices FROM HV through the esp sw.

The following talks about  about the command flow from HV to the esp-MQTT. Harry's attachment in his original email to Martin shows a possible command flow in the other direction.

As Martin points out, receiving commands from HV might require additional code on his part to add HV specific ones to the CLI. Since these commands don't actually exist right now, we have a lot of flexibility on what the syntax is. I'll suggest some here, but the details can be different to accommodate ease of implementation in the esp.

Essentially,  we need to pass a topic and payload.

Some background:  I've attached a very crude system diagram to help visualize how things are/would be connected.  The way things work with the current system - a MQTT client plug-in running on a PC connected to the main serial port of the HV controller - is that the HV controller program sends a serial string to the plug-in which then publishes a topic/payload based on the contents of the serial string it received.  For example, the HV program might send something like this:

    mqtt: livingroom on;

The plug-in has some additional knowledge (specified in a plug-in configuration) of what the full topic should be (i.e., the Tasmota command structure), so it actually publishes something like this:

   topic: cmnd/livingroom/POWER   payload: on

I suspect that, to minimize non-volatile storage requirements in the esp, we may not want to embed this type of partial topic info in there (unless there is some easy way to do it via the scripting language). So that means we would probably need the HV, via its second serial port, to send a command with the full topic as well as payload. Example:

    publish cmnd/livingroom/POWER  on

HV can terminate the command with carriage return/linefeed, or other characters/binary bytes if necessary.
There is no acknowledgment expected by the HV controller.

Fortunately, I think that sending and receiving can be totally independent.

I don't know what the capabilities are of the esp-MQTT sw or the scripting language, so Martin is in the best position to help determine the command structure.

Regards,

Ron

martin-ger, 2017-11-29:
Actually, there is already a command on the CLI. If you send:
publish local <topic> <data>
this will be published to the broker and any subscribed client will revceive this topic.

Caveats are: currently the serial interface is always configured for 115200 bps and it will print some infos (warnings, errors) during operation (not relevant as long as you only need this direction).

rebel7580, 2017-11-30:
We'd need to drop the speed down to 19200. Possible?

martin-ger, 2017-11-30:
No problem, will make that a config parameter.

martin-ger, 2017-11-30:
Done - on the CLI use:
`set bitrate [bps]`
to change the serial bitrate (default is still 115200) and
`set system_output 0`
to stop any info/warning outout from the system. CLI input and acks are still printed as well as script "print"s.

In a script you can enforce these settings e.g. with
```
% Config params, overwrite any previous settings from the commandline
config bitrate	19200
config system_output	0
```

martin-ger, 2017-11-30:
Please note that this refers to the new project at: https://github.com/martin-ger/esp_mqtt

rebel7580, 2017-11-30:
Great! Thanks for the quick response!!

martin-ger, 2017-12-01:
Added some more features to allow for full handling of the serial interface in a script. If you can define for both directions:
- HV--> ESP: which serial output should result in which MQTT message pub and
- HV<--ESP: which received MQTT message should lead to which serial input

it should be reasonable easy to write this script.

rebel7580, 2017-12-01:
HV to ESP:
 For this direcftion, I think we can use the existing publish command. The HV sw would issue a publish command when it wants to turn on/off a device. Ex.:

   To do this:    publish local cmnd/livingroom/POWER on

 HV would execute this code:
  Serial port 3 : Transmit string 'publish local cmnd/livingroom/POWER on'
  Serial port 3 : Transmit Carriage return and Line feed

The ESP should see  "publish local cmnd/livingroom/POWER on\r\n"    \r\n -> 0x0d 0x0a
The HV expects no response.

ESP -> HV:
For this direction, it looks like a simple script can do what we need.
A script receives  on or off command msg and sets the appropriate flag. I think it should look something like this:

on init
do
    subscribe local stat/livingroom/POWER
    subscribe local stat/bedroom/POWER
    ... (additional subscribes, one for each of the remaining devices)

% livingroom -> flag 10
on topic local stat/livingroom/POWER
do
    if $this_data = "on" then
        println ",>0A01"
    else
        if $this_data = "off" then
            println ",>0A00"
        endif
    endif
   
% bedroom -> flag 11
on topic local stat/bedroom/POWER
do
    if $this_data = "on" then
        println ",>0B01"
    else
        if $this_data = "off" then
            println ",>0B00"
        endif
    endif

... (additional similar on topics for each remaining device)

There's probably more that should be included in a real script, esp. in init, like usng your new set baud rate, stop output, etc. , but this should show feasibility.
The HV is expecting the string to be terminated by a carriage return (0x0d). I'm assuming that println will do that.
The HV will not generate any response to these messages

I have some minor concern about flow control (since there isn't any in this proposal), but I think we should only worry about that if it becomes a problem. For example, once a message is received and terminated, the HV will run its event to process the message. Until that processing is complete, any additional incoming characters (e.g., a new message coming close after) will be ignored.

rebel7580, 2017-12-02:
Just re-reading the scripting readme, I noticed that the "serial" event seems to capture any incoming command, instead of the command going to the CLI processor, if "system_output" is set to 0, as you recommended above. So that means we would have to add in a "on serial" command and then parse the string to create the publish command we'd need? If so, that means we could shorten the command we send a little, maybe remove "local", shorten "publish" to "pub".

martin-ger, 2017-12-02:
Yes, that's possible.

rebel7580, 2017-12-02:
I want to make sure I'm clear on this:

If we have system_output = 0, then NO CLI commands are valid via the serial port, and we MUST write an "on serial" event to capture our incoming (from HV) "publish" command.

Will the web console at port 7777 of the AP still work if we need to access the Console?

Also, the incoming char string will terminate in CR/LF (0d 0a). The serial event description in the scripting ReadMe say the incoming strings need to be terminated with a CR. What char(s) do you actually mean here? I think we can send whatever is needed, but the CR/LF is easiest.
Thanks!

martin-ger, 2017-12-02:
Short answer: yes, all will go to 'on serial', port 7777 still works and 
serial input ist terminated with CR (0x0D).

rebel7580, 2017-12-03:
OK, Thanks1  We'll terminate our incoming strings with just a 0x0D.

hjahend, 2017-12-04:
Hello Martin,
I tried tot flash an esp12e  via my Win10 PC via an USB link (on both sides).
Used download tool : https://espressif.com/en/support/download/other-tools
Binaries: https://github.com/martin-ger/esp_mqtt/tree/master/firmware
Flashing seems to complete OK (led on esp blinks during the process).
But after the subsequent  ‘power off/on of the esp’ no luck with the terminal link (used atom, Arduino ide and putty in vain). See the attachments for example output.
What is your advise?
Kind regards
Harry

martin-ger, 2017-12-04:
Sorry, don't see any attachment, but maybe this helps:

Sometimes, especially on old ESP-01s, there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/bin/esp_init_data_default.bin?raw=true and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

hjahend, 2017-12-05:
Hello Martin

Also included the bin file at

https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/bin/esp_init_data_default.bin?raw=true
on 0x3fc000 for 4 MB modules ( ESP-12E) in my download, so 3 files in total.

But no luck so far. See attachment of download and Atom terminal. I also checked with Arduino ide and Putty and not better.

What could be wrong? Before flashing the unit worked OK with Nodemcu.

Regards

Harry

martin-ger, 2017-12-05:
Sorry, still don't see the attachment. Could you post it Here as inline text?

hjahend, 2017-12-05:
Now inserted pictures. hope this helps.

Harry

[cid:d39c7b9e-752f-4de9-a9dd-b83535cfbce7]
[cid:8f5eed7d-b208-4d60-a7cf-be54b76081f5][cid:e2343aa3-79d0-4d17-b928-792352f9eabd]

hjahend, 2017-12-16:
Hello Martin,
I tried tot flash an esp12e  via my Win10 PC via an USB link (on both sides).
Used download tool : https://espressif.com/en/support/download/other-tools
Binaries: https://github.com/martin-ger/esp_mqtt/tree/master/firmware
Flashing seems to complete OK (led on esp blinks during the process).
But after the subsequent  ‘power off/on of the esp’ no luck with the terminal link (used atom, Arduino ide and putty in vain). See the attachments for example output.

What is your advise to proceed?

Kind regards

Harry 
[image]
[image]
[image]

martin-ger, 2017-12-17:
Flash params look fine. Please try to verify the SHA1 checksums of the firmware files. Especially on Windows you might have corruption of binary files caused by the CR/LF problem when downloading. Did you download the firmware files separately? If so. try to download the complete zip-file and extract them. Any difference?

hjahend, 2017-12-17:
Martin,
Dowloaded as zip, SHA1 checked + tried several times to flash the bin files, but still no luck. 
Non performance exactly as before.
So I checked (again) the correct functioning of the ESP 12E 8266 module  with other scetches.
With Arduino-IDE I compiled and downloaded to the esp :
- a blink scetch

- Wifi ScanNetworks scetch.
No problems at all with this action and both worked OK.

After these successful tests my perception is also that  the download to the esp module of the two esp-mqtt bin files via the downloadtool : https://espressif.com/en/support/download/other-tools esp takes far less time to complete than the two above mentioned scetches. Could that   indicate that the code is not completely downloaded to the Esp and if so what next?

Since compiling and downloading via Arduino IDE is already working fine (see above) my question to you: is it possible to have:
a) a complete scetch that I can compile and download with Arduino IDE, or
b)  a bin file compatible with Arduino IDE (import it) and download that to the esp?
c) or use Atom to do the job via importing  a complete scetch + compiling + downloading to esp
If that works well the cause of the previous problems is related to the download tool usage/performance and we do have a successful alternative.

Help is very much appreciated,
thx Harry

rebel7580, 2017-12-17:
Harry -
Did you try DIO instead of QIO mode to flash? (I don't know what the difference is, but Martin suggested this in his ReadMe, so I thought I'd mention that.)
Ron

hjahend, 2017-12-17:
I did QIO

Harry

hjahend, 2017-12-19:
Martin, Ron,
Also tried DIO.  After downloading the bin files as usual, disconnecting and reconnecting the USB link  PC-ESP the blue led on the ESP is blinking like crazy and on a connected terminal I see a lot of symbols passing. Not working properly but apparently running wild. No way to stop it except power off.
Please suggestions.
Thx
Harry

martin-ger, 2017-12-19:
This is the typical behaviour with a non-matching version of "esp_init_data_default.bin" in the flash. See above:

"If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/bin/esp_init_data_default.bin?raw=true and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1)."

Did you check this?

hjahend, 2017-12-19:
Will try that asp, but could not reach that info; link seems broken. Alternative link?
thx
Harry

loadedtech, 2017-12-28:
For what it's worth I'm having the same problem and the link isn't working for me either.  Anyway clues / tips?

martin-ger, 2017-12-28:
Try this one: https://github.com/espressif/ESP8266_NONOS_SDK/raw/master/bin/esp_init_data_default_v08.bin

loadedtech, 2017-12-28:
thanks Martin, but still no joy:
 - MyAP doesn't appear
 - gobbleygook in the serial monitor - first line: {ll|l|  

No errors reported throughout the process and checksums seem ok.   FYI, I'm using 
* wemos nodemcu devkit board 4MB
* esptool v2.0-beta2  
* Mac OS X Sierra

Steps followed:
1.  erase flash
2. write firmware/0x00000.bin 0x10000 firmware/0x10000.bin
3. write esp_init_data_default_v08.bin to 0x3fc000

esptool.py --port /dev/cu.wchusbserial1450 erase_flash
> esptool.py v2.0-beta2
> Connecting....
> Detecting chip type... ESP8266
> Uploading stub...
> Running stub...
> Stub running...
> Erasing flash (this may take a while)...
> Chip erase completed successfully in 24.8s
> Hard resetting...

esptool.py --port /dev/cu.wchusbserial1450 write_flash -fs detect 0x00000 firmware/0x00000.bin 0x10000 firmware/0x10000.bin
> esptool.py v2.0-beta2
> Connecting....
> Detecting chip type... ESP8266
> Uploading stub...
> Running stub...
> Stub running...
> Attaching SPI flash...
> Configuring flash size...
> Auto-detected Flash size: 4MB
> Flash params set to 0x0040
> Compressed 41024 bytes to 27176...
> Wrote 41024 bytes (27176 compressed) at 0x00000000 in 2.4 seconds (effective 134.8 kbit/s)...
> Hash of data verified.
> Compressed 246420 bytes to 170045...
> Wrote 246420 bytes (170045 compressed) at 0x00010000 in 15.2 seconds (effective 129.7 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting...

esptool.py --port /dev/cu.wchusbserial1450 write_flash -fs detect 0x3fc000 esp_init_data_default_v08.bin
> esptool.py v2.0-beta2
> Connecting....
> Detecting chip type... ESP8266
> Uploading stub...
> Running stub...
> Stub running...
> Attaching SPI flash...
> Configuring flash size...
> Auto-detected Flash size: 4MB
> Compressed 128 bytes to 75...
> Wrote 128 bytes (75 compressed) at 0x003fc000 in 0.0 seconds (effective 31.5 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting...

Anyone tips gratefully accepted!

martin-ger, 2017-12-28:
Please try --flash_mode dio first.

Also, havn't seen that compression stuff before...? You might try with another flasher like the one I gave in the link.

loadedtech, 2017-12-28:
Martin, dio did the trick!  Thank you.  Compression was still enabled and seems ok.

For the record, here are the commands I used
esptool.py --port /dev/cu.wchusbserial1450 erase_flash
esptool.py --port /dev/cu.wchusbserial1450 write_flash -fm dio -fs detect 0x00000 firmware/0x00000.bin 0x10000 firmware/0x10000.bin
esptool.py --port /dev/cu.wchusbserial1450 write_flash -fm dio -fs detect 0x3fc000  esp_init_data_default_v08.bin

Now I look forward to playing with your project.  Happy New Year, Hugh

hjahend, 2018-01-02:
Hello Martin,
Happy New Year. Thx for the advise: "If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/bin/esp_init_data_default.bin?raw=true 
Adding that to the two binary files and downloading them all together in dio mode via usb to my ESP 8266 12E got it working. I tested and reconfigured it via an Arduino IDE serial connection from my Win 10 PC.  The soft AP (MyAP) appeared on my scanner (Android). Via ‘scan’ command on the Arduine IDE terminal  I could also find other networks to.
But to my regret I could not connect tot MyAP via IP in a browser and also not via my Sonoff ESP device(s) so far (they have  Tasmota software on them with mqtt capabilities enabled: host = MyASP and port= 1883). See the printout of the Sonoff unit info below.
Maybe I still have to configure something on MyAP and/or on the Sonoff unit?
Another issue is that after a few 'power of of the ESP 12E with MyASP on it, the Wifi connection of it  remains in a try situation (led blinking now an then. May be a config issue; should it be locked to be sure?
Appreciate your suggestions very much.

[image]
[image]

hjahend, 2018-01-29:
Hello Martin,
I am very happy with your ESP-mqtt solution in combination with the Homevision Controller (HV) and Ron  Bostons HV plugin for mqtt. The next challenge is a direct serial link between the ESP and the Homevision Controller (commands and feedback). The script below is intended for feedback from ESP to HV. 
But so far I could not load it in the ESP  via a connected Arduino Ide on a Win 10 PC. 
1. Tried  first to use  the web browser link of HV where the scrip resided:
CMD>script  http://192.168.1.6:11080/script.txt
HTTP request to http://192.168.1.6:11080/script.txt started
CMD>
2. Secondly a download from Github (same script, facilitated by Ron Boston):
CMD>script https://raw.githubusercontent.com/........../script.txt
HTTP request to https://raw.githubusercontent.com/........./script.txt started
CMD>client handshake start.
client handshake ok!
client's data invalid protocol
Error: SSL error 3
=====
Can you help Ron Boston and me out on the following issues:
A: a way to get the script correctly loaded into the ESP via the Arduino ide
B: review the script on possible mistakes that prevent it form loading/functioning.
Kind greetings
Harry Hendriks

script.txt:

% Config params, overwrite any previous settings from the commandline
% config bitrate  19200
% config system_output    0

on init
% Subscribe
do
    subscribe local stat/Woonkmr/POWER
    subscribe local stat/Buiten/POWER
    subscribe local stat/SVTest/POWER
%    ... (additional subscribes, one for each of the remaining devices)

% Woonkmr -> flag 32
on topic local stat/Woonkamer/POWER
do
    if $this_data = "on" then
        println ",>2001"
    else
        if $this_data = "off" then
            println ",>2000"
        endif
    endif

% Buiten -> flag 33
on topic local stat/Buiten/POWER
do
    if $this_data = "on" then
        println ",>2101"
    else
        if $this_data = "off" then
            println ",>2100"
        endif
    endif

% SVTest -> flag 34
on topic local stat/SVTest/POWER
do
    if $this_data = "on" then
        println ",>2201"
    else
        if $this_data = "off" then
            println ",>2200"
        endif
    endif

%... (additional similar "on topic"s for each remaining device)

on serial
do
    % parse inputs of the form: pub,{topic},{payload}
    if csvstr($this_serial,0,",") = "pub" then
        publish local csvstr($this_serial,1,",") csvstr($this_serial,2,",")
    endif

martin-ger, 2018-01-29:
Sorry, a recent change to add http requests in scripts has broken the script download via http. This is fixed in the lastest upload.

So, upload the script to a web server and then load it from the ESP using "script your_url" as you tried already.

I checked the script, nearly perfect, i think. Made only some minor fixes and it seems to work for (as far as I can test without a HV):
```
% Config params, overwrite any previous settings from the commandline
% config bitrate 19200
% config system_output 0
on init
% Subscribe
do
subscribe local stat/Woonkmr/POWER
subscribe local stat/Buiten/POWER
subscribe local stat/SVTest/POWER
% ... (additional subscribes, one for each of the remaining devices)
% Woonkmr -> flag 32

on topic local stat/Woonkamer/POWER
do
	if $this_data = "on" then
		serial_out ",>2001" | #0d
	else
		if $this_data = "off" then
			serial_out ",>2000" | #0d
		endif
	endif

% Buiten -> flag 33
on topic local stat/Buiten/POWER
do
	if $this_data = "on" then
		serial_out ",>2101" | #0d
	else
		if $this_data = "off" then
			serial_out ",>2100" | #0d
		endif
	endif

% SVTest -> flag 34
on topic local stat/SVTest/POWER
do
	if $this_data = "on" then
		serial_out ",>2201" | #0d
	else
		if $this_data = "off" then
			serial_out ",>2200" | #0d
		endif
	endif

%... (additional similar "on topic"s for each remaining device)

on serial
do
	% parse inputs of the form: pub,{topic},{payload}
	setvar $topic_str = csvstr($this_serial,1,",")
	setvar $data_str = csvstr($this_serial,2,",")
	if csvstr($this_serial,0,",") = "pub" then
		publish local $topic_str $data_str
	endif
```

hjahend, 2018-01-31:
Thank you very much Martin.

Downloaded the complete set from github and tried to flash the bin files via the ESP tool for Windows to the ESP826612E.

Encountered problem: accessing the  0x10000.bin file denied. Other bin file(s): no problem to download them in the ESP.

Please advise a solution.

Kind greetings

Harry

[image]

martin-ger, 2018-01-31:
If you download 0x10000.bin separatly from github?

rebel7580, 2018-01-31:
Finally got to trying this code myself today. I was able to flash (separately ) both of the .bin files without any problem, as well as the third one recommended in the thread. Here's what I did (using an ESP8266-01s):

esptool.py --port COM5 erase_flash
esptool.py --port COM5 write_flash -fm dio -fs 1MB  0x0 0x00000.bin
esptool.py --port COM5 write_flash -fm dio -fs 1MB  0x10000 0x10000.bin
esptool.py --port COM5 write_flash -fm dio -fs 1MB  0x3fc00 esp_init_data_default_v08.bin

These all worked fine, as far as the downloads are concerned.

However, the code does not seem to be working, or I'm doing something wrong myself. I have no serial connection, nor do I see the MyAP wifi.

But at least I didn't have any issue with the flashing itself...  :)

Ron

martin-ger, 2018-01-31:
For an ESP-01 with 1MB of flash esp_init_data_default_v08.bin has to be flashed to address 0xFC000. Guess, this is the problem.

rebel7580, 2018-01-31:
Ahh, of course!  Thanks.

rebel7580, 2018-02-01:
That worked.  Got my ESP up and running, and was able to load the script. Of course, this particular script won'r work for me, but it shows that I could load it via the HVXL webserver.
Thanks.

martin-ger, 2018-02-01:
I tested the script exactly in the described config: "system_output 0", console connected via telnet 7777. Then, after init only serial_out goes to the UART. But then your script should NOT print the ",>2201" to the console but only to serial. It does for me...

An input string on the serial has to be termininated with an CR character, try to enter CRTL-M after the pub command.

rebel7580, 2018-02-01:
Then I must have something else set wrong. I keep getting serial_out sent to the console, not the serial port.
 For sending out through the serial port, I tried CTRL-M, but no success there either.

I've restarted several times, but same thing every time.  I'm at a loss to explain why it works for you but not me!

rebel7580, 2018-02-01:
Here's my complete script. The print/println are there for debugging. I change system_output manually via the console. then save/reset
```
% Config params, overwrite any previous settings from the commandline
% config bitrate 19200
% config system_output 0

on init
% Subscribe
do
subscribe local stat/Woonkmr/POWER
subscribe local stat/Buiten/POWER
subscribe local stat/SVTest/POWER

% ... (additional subscribes, one for each of the remaining devices)
% Woonkmr -> flag 32

on topic local stat/Woonkamer/POWER
do
    if $this_data = "on" then
        serial_out ",>2001" | #0d
    else
        if $this_data = "off" then
            serial_out ",>2000" | #0d
        endif
    endif

% Buiten -> flag 33
on topic local stat/Buiten/POWER
do
    if $this_data = "on" then
        serial_out ",>2101" | #0d
    else
        if $this_data = "off" then
            serial_out ",>2100" | #0d
        endif
    endif

% SVTest -> flag 34
on topic local stat/SVTest/POWER
do
    print "SVTest: "
    println $this_data
    if $this_data = "on" then
        serial_out ",>2201" | #0d
    else
        if $this_data = "off" then
            serial_out ",>2200" | #0d
        endif
    endif

%... (additional similar "on topic"s for each remaining device)

on serial
do
    % parse inputs of the form: pub,{topic},{payload}
    setvar $topic_str = csvstr($this_serial,1,",")
    setvar $data_str = csvstr($this_serial,2,",")
    if csvstr($this_serial,0,",") = "pub" then
        publish local $topic_str $data_str
    endif
    println $this_serial

```

rebel7580, 2018-02-01:
OK, I played a little with the serial terminal settings in PuTTY and now I can send out through the serial port!
I successfully sent a "pub" message!

But  serial_out still comes out on the console.
In case I had some other setting that was blocking serial output, I checked the FTDI I use to connect the ESP to my PC. It blinks whenever I type IN chars to send ("pub" etc.), but when I should be getting something out, no blinking. I interpret that to mean that the ESP isn't sending anything out the serial port.

So close....

Ron

martin-ger, 2018-02-01:
Ok, see the problem: If publish to "stat/Buiten/POWER" it works, if I do it to "stat/SVTest/POWER" not. Obviously, the println before disturbes that. Should not happen and I will look into that.

But for now, it should be enough to remove the debug output. Does this work for you as well?

Publishing also works for me: blindly typing "pub,1,2<CTRL-M>" on the serial gives me a publication (with the trailing CR in the data, but I will remove that in the next version)

rebel7580, 2018-02-01:
Yes! Now I think we have it!!!!  Not sure why I didn't see the received data before I put in those print/println lines. I put them there because I wasn't seeing anything, but I guess "user error" at that point. :)

In my client, I do notice that extra character in the published message.

Thanks for you help and patience.

Really going to sleep now...

Ron

martin-ger, 2018-02-01:
Good night! ;-)

Think, I fixed the two issues now:
- serial input waits for <NL> terminated strings,(it is actually <CR>), cuts of the last char, and hands it over to the script
-  serial_out should not any more mix up with prints

[12 more replies]

closed 2018-02-09 by martin-ger