martin-ger/esp_wifi_repeater #45: Hide ESP SSID

dindan100, 2017-05-31:
Dear MR martin,

Just wanted to say thanks for the time and effort you have invested in writing this code and the documentation which goes with it!!!
Everything is crystal clear even for a newbie like me and although i failed to install it using the predefined bin i have followed your remarks for the ESP01 issue and using the other bins which worked just fine (i am not using ESP01 but ESP12E with a 512KB flash memory but i guess this is reason for failure).

I also wanted to ask if it possible to hide the ESP SSID from the outside world as a basic security measure as i am extending my network coverage using your wonderful work i do not want people to scan my network and try to hack it.

Thanks again and god bless you!

martin-ger, 2017-06-01:
Actually, this was quite easy to add - find it in the latest version ("set ssid_hidden 1"). Thank you for the idea!

dindan100, 2017-06-01:
Thanks buddy,

will try to test it today and let you know if i have another ideas you might want to consider :)

closed 2017-06-01 by martin-ger

marek26340, 2017-06-03:
Well, i think for the command, this should be better: set **ap_**ssid_hidden