martin-ger/esp_wifi_repeater #46: wifi repeater is not working after upgrade

dindan100, 2017-06-03:
Hi,

I am using ESP12E with 512kb of memory.
I have installed successfully the previous version from last week and it was working amazingly but since i have installed the new version from the last days over the older version  to check the new SSID_HIDDEN feature the wifi repeater is not working any more.
The "bpllblrl▒r▒n▒b▒nbp" string apears upon reboot but the CMD is not responsive hence cannot be configured or used.

* I am using the files in the older SDK directory using the 0x00000 and 0x40000 address as before
* There's no power issue here.
* Any example from the arduino examples is uploaded safely and working just fine.
* i have used the esptool.py -p com7 -b 115200 erase_flash to try and clean the flash but it is not working.

Any word of advise??

Thanks in advance.

martin-ger, 2017-06-04:
Just double-checked the latest firmware (both SDK 1.5.4 and 2.0) on several ESP-platforms (ESP-01, Wemos D1 mini, ESP-12). I experienced no problems.What platform do you use? Still able to run the older FW?

dindan100, 2017-06-04:
Hi Martin,

i am using ESP12E with 512KB.
I am using win7 to upload the bin files to the ESP.
No, i cannot use the older version nor the new version anymore as although the upload always turn to be successful when booting the device there no response.
I have tried using adruino examples and even installed easy esp on this device and they are all working fine so it is very strange.

martin-ger, 2017-06-04:
The SDK 2.0 firmware should be okay for you. Do you see the "WiFi Repeater V1.4 starting" string on reboot? If not, there is something wrong with the flashed image.

dindan100, 2017-06-04:
No, i dont see the "WiFi Repeater V1.4 starting" string upon reboot which leads me to the conclusion there is something wrong with the flashed image but not with the image itself.
As for the SDK 2.0 image ,last time i used  it the flash succeeded but upon reboot there was infinite reboot so i had to use the 1.5.4 version which worked great.
This time using both images is the same meaning no response from the CMD.

dindan100, 2017-06-04:
Just tried to use the old bins and the new ones on another esp chip (ESP201-512KB).
Same behavior meaning bins from arduino and easyeasp are working but not the wifi repeater bin

dindan100, 2017-06-04:
My Dear friend,

I have finally found the problem!!! and it has nothing to  do with your bin file as expected!!
After many and painful hours i have found that when you download the bin files using firefox V53.0.3 there are 4 extra bytes in the bin file compared to downloading with chrome!!!!! dont know how or why but this is a fact (at least on my computer).
I used the bin files downloaded with the chrome and walla everything is working just fine (at least the 1.5.4 SDK files).
I also tested the new SSID_hidden flag and it working great! (my phone failed to detect the network as expected but using manual configuration to configure the new "router" works just fine!

I suggest 2 small enhancements:
1. Add the cksum of the bin files you have compiled and add to the instructions to check the cksum after downloading the file so other wont suffer as i did :)
2. Add the current SSID_HIDDEN status to the show and show stats commands so when pepole will forget that they enabled this feature and  wonder why they cant find the network there will be an indiction why.

Thanks for your time and wiliness to assist buddy!

closed 2017-06-04 by dindan100