martin-ger/esp_wifi_repeater #48: question ssid and or feature

Geekynooby, 2017-06-09:
Hi iam new to all this.. but you did a fine tutorial and work:)

I have two questions. 
Is it possible if you set ssid to ad a space in middler of the ssid ? What happens now is ( set ssid my wifi ) for example, after save and reset it only tries to connect to "my" instead of "my wifi " cant find no solution so far on the web, or is this impossible ? 

Other question more a feature: Wpa Enterprise with WPA/WPA2, EAP-PEAP, MSCHAPV2, for networks with and without certificate and user name and password? Would be neat but  think a little difficult.

Keep up the good stuff

thx

martin-ger, 2017-06-10:
Hi,

good news: both is already done: spaces in the SSID can be quotet using the '\' character. For "my wifi" write:
`set ssid my\ wifi`

WPA Enterprise have a look into https://github.com/martin-ger/esp_peap_psk 
It is a wild hack, but currently the only way to do it, if the outer ID string is not ignored from the network.

Geekynooby, 2017-06-11:
Whoow thats fast:) 

Ty very much gonne look it up 

Tia

closed 2017-06-11 by Geekynooby