martin-ger/esp_wifi_repeater #49: Forward port from broadcast IP UDP

TienHuyIoT, 2017-06-13:
I want to forward port from broadcast IP UDP.
Ex: 255.255.255.255:12345 -> 10.0.1.2:12346
May you tell me how to do it. Because I want to find IP device by UDP broadcast and esp wifi repeater.
Tks very much Martin-Ger!

martin-ger, 2017-06-13:
Should work for BC as well:
portmap add 12345 10.0.1.2 12345

TienHuyIoT, 2017-06-14:
I have 4 IP:
1. ESP connect to STA(Home Wifi) get IP: 192.168.1.50
2. My Laptop connect to STA(Home Wifi) get IP: 192.168.1.59
3. My phone connect to ESP AP have IP: 10.0.1.2
4. ESP AP gateway IP: 10.0.1.1
I have set sucess Comand "set portmap add UDP 12345 10.0.1.2 12345". And test UDP unicast OK when send data form IP 192.168.1.49:25123 to 192.168.1.50:12345. This time, Esp will portmap 192.168.1.50:12345 to 10.0.1.2:12345 OK.
But when i try UDP IP broadcast send data from IP 192.168.1.49:25123 to 255.255.255.255:12345. This time, Esp will not portmap 192.168.1.50:12345 to 10.0.1.2:12345 OK.
May you resolve this issues?

martin-ger, 2017-06-14:
Ok, thought about it: this is actually not a bug, it is the spec: 255.255.255.255 is a BC at network scope - these BCs are not routed into other networks (for good reasons). That's why it is also not forwarded. 

Will have a look into the lwip whether we can overwrite that in the NAT case, but I guess it remains a violation of the IP specs...

TienHuyIoT, 2017-06-14:
May you update NAT case. I think it useful when we need to find an IP ESP in network by UDP broadcast.
Now, i find esp Ip by send string "Who are ESPs?". If have ESP in network. They'll feedback to ...

TienHuyIoT, 2017-06-15:
May you make a video how to buil this project in windows. Because don't know buil in linux.
Tks so much!

ShinDaiIchi, 2017-07-02:
How to change ESP AP gateway from default AP IP:192.168.4.1 to AP IP: 10.0.1.1?

martin-ger, 2017-07-02:
set network ip-addr: sets the IP address of the internal network, network is always /24, router is always x.x.x.1

ShinDaiIchi, 2017-07-02:
Thank you.  # @martin-ger