martin-ger/esp_wifi_repeater #51: Unable to connect

Matthww, 2017-06-18:
Hello

I’m unable to connect to the ESP after I configured everything with telenet. The ESP is visible on the router it’s connecting too and I can connect to the ESP from ththe router wifi but when I try to connect to the ESP ap it gives me a error saying “unable to connect”

martin-ger, 2017-06-18:
Could you send me the output of "show" and "show stats"?

Matthww, 2017-06-21:
show:
```
STA: SSID:almerewireless PW:pass
AP:  SSID:Matthew  PW:matthew1234test [open] IP:192.168.4.1/24
STA MAC: 5c:cf:7f:priv:priv:priv
AP MAC:  5e:cf:7f:priv:priv:priv
Clock speed: 80
MQTT: disabled
```

show stats:
```
System uptime: 0:21:18
Power supply: 2.714 V
0 KiB in (0 packets)
0 KiB out (0 packets)
External IP-address: 192.168.1.208
0 Stations connected to AP
Station: 9c:f3:87:priv:priv:priv - 192.168.4.2
```

Matthww, 2017-06-21:
I tried to reflash it but now it's asking for a password to connect to an open wifi network??

martin-ger, 2017-06-21:
Reflashing doesn't change the stored settings, try to do "reset factory" instead.

Your log shows, that it tried to connect (and maybe failed). However your ESP's AP is open, i.e. the password will not be used.

TienHuyIoT, 2017-06-28:
Try flash an example SoftAP to Your ESP with ap_ssid and ap_password.
Then, flash esp_repeater will OK.

Toxlen, 2020-06-23:
I don't know if I should open an other issue or not but hear is my problem :
I have a similar issue but i can connect and it works when the ap_open is 1.
Just when there is a password and the ap_open is 0 my computer can't connect.

```
station: 34:02:86:priv:priv:priv join, AID = 1
station: 34:02:86:priv:priv:priv join, AID = 1
E:M 672
E:M 672
E:M 672
E:M 672
E:M 672
station: 34:02:86:priv:priv:priv leave, AID = 1
rm 1
station: 34:02:86:priv:priv:priv leave, AID = 1
```

martin-ger, 2020-06-23:
"E:M XXX" is a typical error messages of the driver when it has a lack of buffers. It is an indication of overload. I am not aware how to improve that.