martin-ger/esp_wifi_repeater #52: Multiple devices P2P

diffstorm, 2017-06-18:
Hi,
I have a question about capability of the system.
For example; can I connect 1000 devices one by one to repeat the internet to long range?
modem-->esp0-->esp1-->2-->3-->4-->5-->6-->...........1000-->user
Can user have internet connection even if it is very slow?
BR

martin-ger, 2017-06-18:
In theory, yes, if you make sure that the network addresses at each router are different (however you can reuse addresses at different levels). 

Practically, for each repeating process the frame is send once more over the same WiFi channel. This will saturate the channel quite soon. So, 1, 2, 3, and even 4 hops are known to work. Anything (far) beyond will probably end up in massive message loss...

Building a kind of a distribution tree around an area of several 100m might work (and I already thought about some way of autoconfiguration for that). But it will fail for a nationwide carrier... ;-)

diffstorm, 2017-06-18:
Thanks for the answer.

WPS would be nice for some automatic connection even between 2 ESP repeaters.
For example it would add a number to STA SSID (e.g SSID2, SSID3) automatically, so user can configure it without writing console commands.

diffstorm, 2017-11-25:
I see that actually you have implemented a feature what I was asking for.
What if I connect 1000 esp-s in a row (lets say 50 meters per esp)?
Would it work nicely?
If yes, every esp can have the internet access and if we write our own code, we can talk to the server directly over internet, right?

martin-ger, 2017-11-25:
As said before: 1000 won't work. Too many hops, too much losses. 10 will probably work, but slow...

Yes, you have access to the internet from all levels, while connection setup from the internet to the clients will fail. The usual NAT problem. (port redirects have to be set up manually)

diffstorm, 2017-11-25:
Do you determine the hop number/level automatically or user should write it using a command?
if it's an automatic process, how did you do that?
if not - any idea to make it automatic? so we will program them using hard coded SSID-PSK and they will determine which hop they are for each.

martin-ger, 2017-11-26:
Automatically: each esp_wifi_repeater configured with "automesh" will first search for the best other AP to connect to. This is the one which closest to the original WiFi network and has the best signal strength (RSSI).  But which is the one closest to the original WiFi network if you see several APs with the same SSID? Therefore I use a somewhat dirty trick: the esp_wifi_repeaters in "automesh" mode manipulate their BSSID, i.e. the MAC address of their AP interface which is send out with every beacon frame 10 about time times per second. It uses the format: 24:24:mm:rr:rr:rr. "24:24" is just the unique identifier of a repeater. "mm" means the "mesh level", this is the distance in hops to the original WiFi network. The last three "rr:rr:rr" are just random numbers to distinguish the various ESPs.

Now each esp_wifi_repeater can learn which other esp_wifi_repeater is the closest to the the original WiFi network, can connect to that, and chose its own BSSID accordingly. Also the IP address of the internal network is adjusted to the mesh level: 10.24.m.0. This creates a tree (a very special mesh) with the original WiFi network as root and nodes on several mesh levels (actually, it works somewhat similar as the Spanning Tree Protocol (STP) on the link layer or routing on the network layer using a Distance Vector protocol). As soon as a link loss is detected, configuration is restarted. This should avoid loops as during (re-)configuration also no beacons with an BSSID are sent.

diffstorm, 2017-11-26:
When power is applied to all esp-s at the same time, how does act the last esp? First one will find the wifi modem and it will connect, in this time second and all others are waiting till somebody have correct MAC?

martin-ger, 2017-11-26:
All will start scanning for the configured SSID. As only those which are connected to the internet (via the root or any higher level) are sending beacons, the outer ESP will wait until a link is available.

diffstorm, 2017-11-26:
Great solution, but this delays so much if 1000 esp-s in a row. (I know it won't work)

Let me ask another question; for the 1000 esp-s, if I connect one to another and send udp broadcast packets from one end and make sure every esp broadcasts the packet which is received, is this method effected by the loss like your solution or it can work? (of course I will add some conditions to prevent to re-broadcast the same packets continuously)

first send broadcast
second receive and broadcast
third receive and broadcast
Nth receive and broadcast

geofrancis, 2017-11-26:
The problem is you half your bandwidth after every hop since each router is half duplex. Ideally if there was a way of joining 2 routers then you could retransmit on a different channel eliminating the bottlenecks

martin-ger, 2017-11-26:
I think the solution with the broadcasts is about what Espressif does with their mesh. Can work even without a AP-STA association, but is not so easy to prevent circles in the network (and infinite message replication). You have to remeber messages you send before - difficult if many stations are allowed to send. Still it will be hard with 1000 nodes, even if you have only 1% probability of message loss (expecially as broadcasts are not acknowledged). (100 hops: loss probability 1-0.99^100 = 63%)

Discussed the idea with two ESPs before. Problem: there is no interface between those two with 54 Mbps.

diffstorm, 2017-11-26:
OK but if the loops are prevented and loss is detected with ack-s, it is possible to work.
Espressif mesh cannot work with 1000 hops.
Please close the topic if you don't have any to add.
Thanks for the brainstorming.

closed 2017-11-26 by martin-ger