martin-ger/esp_wifi_repeater #53: Feature request - custom DNS servers

robdixn, 2017-06-25:
An option to set custom DNS servers for services such as OpenDNS would be ideal. I’m trying to use my nodemcu as an access point for guests and being able to restrict certain bandwidth-hogging websites using OpenDNS would be great!

martin-ger, 2017-06-25:
With "set dns x.x.x.x" you can already define a fixed DNS server for the clients.

robdixn, 2017-06-25:
Thanks. Could I do this for 2 IPs eg. 1.1.1.1 & 1.1.1.2 or just one IP? If I can only use one IP, how would I go about using OpenDNS with this firmware?

martin-ger, 2017-06-25:
It is just one DNS address that is given to the clients (DNS always uses just one, the second is backup in case the first is not accessible). Just try "set dns 208.67.222.222" - should work...

robdixn, 2017-06-25:
Ah ok that makes sense, I’ll give it a go. Thanks for the speedy support btw.

robdixn, 2017-06-26:
I can confirm that OpenDNS is working for anyone who would also like to use it with this firmware. Tried it myself and it’s working perfectly.

closed 2017-06-26 by robdixn