martin-ger/esp_wifi_repeater #54: How to set MQTT-Client?

ShinDaiIchi, 2017-06-28:
I have set MQTT-Broker by mosquitto on 192.168.4.2
and set MQTT-Client as the followiing ,  

`
--------------------------------------------------------
>telnet 192.168.4.1 7777
--------------------------------------------------------
CMD>set mqtt_host  192.168.4.2
MQTT host set

CMD>set mqtt_port  1883
MQTT port set

CMD>set mqtt_command_topic /WiFi/Router01/command
MQTT command topic set

CMD>set mqtt_interval 5
MQTT interval set

CMD>set mqtt_mask ffff
MQTT mask set

CMD>save
CMD>reset
--------------------------------------------------------
`
However when I open MQTT-SPY  , MQTT it seem not work .
How to set MQTT-Client to view?

Thank you.

martin-ger, 2017-06-28:
Config looks good (if 192.168.4.2 is really the MQTT broker, what I assume). What does "show mqtt" report? Is there any message in the serial log at boot time?

ShinDaiIchi, 2017-06-28:
`
CMD> show mqtt
MQTT host: 192.168.4.2
MQTT port: 1883
MQTT user:
MQTT password:
MQTT id:
MQTT prefix:
MQTT command topic: /WiFi/Router01/command
MQTT gpio_out topic:
MQTT interval: 5 s
MQTT mask: ffff

CMD>reset
`
After reset ESP-WiFi-Repeater,  MQTT-Broker on 192.168.4.2  will be disconnected automatically.
And message in the serial log at boot time as the following..

[Serial log]
`
Repeater V1.4 starting

Config found and loaded
Starting Console TCP Server on port 5559
mode : sta(18:fe:34:e0:bf:ff) + softAP(1a:fe:34:e0:bf:ff)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with MyHome, channel 6
dhcp client start...
connect to ssid MyHome, channel 6
ip:192.168.43.99,mask:255.255.255.0,gw:192.168.43.1
ip:192.168.43.99,mask:255.255.255.0,gw:192.168.43.1,dns:192.168.43.1
add 1
aid 1
station: f0:79:59:19:f6:91 join, AID = 1
station: f0:79:59:19:f6:91 join, AID = 1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
LmacRxBlk:1
....
`

And then I reconnect MQTT-Broker (192.168.4.2) to  ESP-WiFi-Repeater.
Can I use this MQTT-Broker to know information?

Thank you.

martin-ger, 2017-06-28:
Config looks okay, but the MQTT-Server doesn't seem to be connected. Could you try to setup the server in the external network (the 192.168.43.x net)? Does it work then? I will try to reproduce your setup later today...

ShinDaiIchi, 2017-06-28:
I change  MQTT-Broker to "iot.eclipse.org"  and port 1883,
however  serial log  seem like internal network , occur line ....

MQTT:Diconnected

martin-ger, 2017-06-28:
You have so many "LmacRxBlk:1", which means buffer shortage, maybe this is a/the problem. Do you have so much traffic during your tests?

martin-ger, 2017-06-28:
Please try: "set mqtt_user none". If you have no username/password auth at the server, it must be explicitly "none".

ShinDaiIchi, 2017-06-28:
I don't know why so many "LmacRxBlk:1", 
But when  "set mqtt_user none"

```
CMD>show mqtt
MQTT host: iot.eclipse.org
MQTT port: 1883
MQTT user: none
MQTT password:
MQTT id:
MQTT prefix:
MQTT command topic: /WiFi/Router01/command
MQTT gpio_out topic:
MQTT interval: 5 s
MQTT mask: ffff

```

Message in serial log doesn't have "LmacRxBlk:1"  any more. as following ..

```
Repeater V1.4 starting

Config found and loaded
Starting Console TCP Server on port 7777
mode : sta(18:fe:34:e0:bf:ff) + softAP(1a:fe:34:e0:bf:ff)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with MyHome, channel 6
ip:192.168.43.99,mask:255.255.255.0,gw:192.168.43.1
connect to ssid MyHome, channel 6
ip:192.168.43.99,mask:255.255.255.0,gw:192.168.43.1,dns:8.8.8.8
MQTT: Disconnected
MQTT: Disconnected
....

```

However still happen  "MQTT: Disconnected"

martin-ger, 2017-06-28:
Please try to set a valid prefix as well. /Test/124/ e.g.

ShinDaiIchi, 2017-06-28:
[Current MQTT setting ]

```
CMD>show mqtt
MQTT host: iot.eclipse.org
MQTT port: 1883
MQTT user: none
MQTT password: none
MQTT id: qqre443434
MQTT prefix: /Test/124/
MQTT command topic: /WiFi/Router01/command
MQTT gpio_out topic:
MQTT interval: 5 s
MQTT mask: ffff
```

[Serial Log after reset]
```
Repeater V1.4 starting

Config found and loaded
Starting Console TCP Server on port 5559
mode : sta(18:fe:34:e0:bf:ff) + softAP(1a:fe:34:e0:bf:ff)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 0 (4)
reconnect
disconnect from ssid MyHome, reason 4
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

connected with MyHome, channel 6
ip:192.168.43.99,mask:255.255.255.0,gw:192.168.43.1
connect to ssid MyHome, channel 6
ip:192.168.43.99,mask:255.255.255.0,gw:192.168.43.1,dns:8.8.8.8
MQTT: Connected

```

It's ok now  ---> "MQTT:Connected"

And then when I use Mqtt-spy  to  the Broker,  
which topic for subcribe I must to set?

ShinDaiIchi, 2017-06-28:
How to set  MQTT gpio_out topic ?  (Don't have in help)

`MQTT gpio_out topic: /WiFi/ESPRouter_d8bcb5/switch`

martin-ger, 2017-06-28:
GPIO_IN and GPIO_OUT are not enabled in the precompiled FW. You have to recompile the sources with changed user_config.h to use this feature.

ShinDaiIchi, 2017-06-29:
I can receive mqtt-message from broker already.

Thank you very much.

closed 2017-06-29 by martin-ger