martin-ger/esp_wifi_repeater #55: Request  NTP automatically timesync.

TridentTD, 2017-06-30:
Wish wifi-repeater has system time sync automatically to NTP-server .

Thank you.

martin-ger, 2017-06-30:
Okay, I could need NTP-real-time in another project - but what would be the requirement in the esp_wifi_repeater other than info?

ShinDaiIchi, 2017-07-01:
When monitoring such as monitor mode or mqtt mode .. etc.
It's maybe useful with gpio in/out too.

ShinDaiIchi, 2017-07-01:
Good figture, I think this figture maybe useful when monitoring 
such as monitor mode or mqtt mode and gpio in/out too.

martin-ger, 2017-07-09:
Added NTP time sync in the "ntp" branch. If you like you might want to test it...

TridentTD, 2017-07-10:
Thank you very much, Martin-ger.

I will try.

TridentTD, 2017-07-10:
After set  NTP server to "ntp.nasa.gov" and restart.
But when querying the NTP time with the "time" ,
I found it still seem not sync the NTP time  ..?

the commandline's response picture ...
https://ibb.co/iFpJkF

Thank you.

martin-ger, 2017-07-10:
"ntp.nasa.gov" seems not to be a valid NTP server (nslookup gives me: "** server can't find ntp.nasa.gov: NXDOMAIN"). Try "1.pool.ntp.org" instead.

TridentTD, 2017-07-10:
I found  some messages occurs many times,
when I try to connect to MyAP at the first time after flashing rom.

This responce message in serial monitoring,
I guess that makes connecting to MyAP is very very hard at the first time.?

>scandone
>no ssid found, reconnect after 1s
>disconnect from ssid ssid, reason 201
>reconnect

Serial's response picture...
https://ibb.co/mXsKua

-----
I will try change to "1.pool.ntp.org"   .
Thank you.

martin-ger, 2017-07-12:
This is debug output from the SDK, not printed from my code. Don't know how the easily switch that off...

closed 2017-07-17 by martin-ger