martin-ger/esp_wifi_repeater #57: Web server

Seshpenguin, 2017-07-08:
Hi,
Is it possible to run a website on the ESP alongside the NAT Router (So people connected to the ESP can access the website)?
Thanks!

martin-ger, 2017-07-08:
Canon be done. The commandline interface can be replaced by a web server. Feel free to start a branch...

Seshpenguin, 2017-07-09:
Do you know how I should go about implementing it?

martin-ger, 2017-07-09:
This could be a good starting point: https://github.com/Spritetm/esphttpd. However, some experiments with the open-sdk would be helpful too.

Seshpenguin, 2017-07-09:
Hm.. I tried a bit but sadly I don't have enough knowledge of low-level C :stuck_out_tongue: 
Maybe someone else can try to implement a webserver??

martin-ger, 2017-07-24:
Added a basic web config interface (Thanks to rubfi).

closed 2017-07-24 by martin-ger