martin-ger/esp_wifi_repeater #58: sim800 support

karioja, 2017-07-09:
Would it be technically feasible to replace the uplink AP connection with a SIM800 module and turn this into a 2g/wifi router? Now since sim800 does not support raw IP something like SLIP over TCP (with socat and slattach on the server) would need to be used.

martin-ger, 2017-07-09:
Technically feasible? Think so: pobably the way to go should be like this: write a new lwip netif (like I did for SLIP over serial in: https://github.com/martin-ger/esp_slip_router ) and replace the STA-interface in this project with this new interface.

So, writing the netif, that handles all the config stuff and finally sends and receives packets is the major challenge. Perhaps anybody has been working in this direction...? (I have an A6 GPRS module here, but hadn't any time so far to look into it)

karioja, 2017-10-27:
https://github.com/loboris/ESP32-PPPOS-EXAMPLE/blob/master/components/pppos/libGSM.c

closed 2019-07-22 by martin-ger