martin-ger/esp_wifi_repeater #61: Feature requests - web management interface and captive portal

robdixn, 2017-07-23:
Been using this to create a guest hotspot for a while (as BT routers do not let you a secondary network for guest access to prevent access to equipment on the primary network) and I love it.

The command line interface for modifying settings is great and all but it would be nice to have a web GUI type interface for modifying settings as well, I believe this is not too hard as I've seen it done [here](https://github.com/spacehuhn/esp8266_deauther).

Not sure if this is actually possible with the esp but it would also be pretty awesome to create user accounts with different permissions/website filtering and have them sign in through a captive portal.

martin-ger, 2017-07-24:
Basic web config is now included...

TridentTD, 2017-07-28:
#define CONFIG_PAGE  in file "web.h",

When I tried to create long  CONFIG_PAGE,  
it can compile , after restart  it's brick!,  

Can use more long #define CONFIG_PAGE?

Thank you.

martin-ger, 2017-07-28:
Currently the whole page is send out with just one tcp_send after connect. To send longer string, we have to change the sending with send partially, send callback, send the nest part and so on. I will look after that after my summer vacation. In  two weeks I will be back... ;-)

robdixn, 2017-07-28:
Would be nice to set the DNS through the web config

TridentTD, 2017-07-29:
@martin-ger  
Have a nice summer vacation. ^^

TridentTD, 2017-07-29:
@robdixn 

Web config with DNS  - ROM  at
https://drive.google.com/file/d/0B2YEwqsbQm08WWIxNWxwYU1oQzQ/view

robdixn, 2017-07-29:
Cheers @TridentTD

robdixn, 2017-07-29:
@TridentTD it's not flashing for some reason, flash process shows up on PC as successful but old SSID is being broadcast with same config options as before. Any ideas what I'm doing wrong?

TridentTD, 2017-07-29:
ESP8266  save the last wifi-config in the flash mem.
Try to clean up the flash by use  esptool.py.

>esptool.py --port [YOUR_COM] erase_flash

robdixn, 2017-07-29:
@TridentTD ok, will try that, what would I use for the '--port' variable?

TridentTD, 2017-07-30:
COMPORT that you connect ESP8266 to your computer.
example COM5  (on Windows)..

>esptool.py --port COM5 erase_flash

ffanez, 2017-07-30:
Martin-ger's master,
Thank you for the repositories you have made.
I've managed to try it, thanks.
But can you help my problem to join the open dhcp wifi network using user login and password?
(Keep using the esp8266 device of course)