martin-ger/esp_wifi_repeater #62: MQTT feature inquery

Muhamed-Reyad, 2017-07-25:
Hi, first i would like to say thank you this works amazingly well with minimum setup required.

I just wanted to ask about the ability to publish a custom MQTT message, e.g. through a UART command, it would be very helpful to get more data sent in a home automation environment paired with an arduino maybe.

Also i can see when viewing "show mqtt" command that there is an "MQTT gpio_out topic" but i couldn't find how to use it in the documentation and whether this is used to control a GPIO pin to switch something or does it read if the GPIO pin is high or not and publish that info to the Broker.

thnx

martin-ger, 2017-07-27:
GPIO_IN and GPIO_OUT are experimental features. Added them when I installed the software on a Sonoff switch. You can enable and define them in user_config.h and adapt it to your own system. As this is a little bit off topic for a router software I decided not to include it to the main line.

Muhamed-Reyad, 2017-07-27:
Great i will try to test them out.
But what about a custom mqtt message ? Do you think it can be implemented easily ?

harryd100, 2017-09-22:
I have been playing around also with GPIO_IN and GPIO_OUT, and it seems t work fine except when using retained messages. The retained message is send by the broker, but the wifi_repeater does not pick it up. I'm using this to have a switch set to the latest state after being disconnected from WiFI, re-connected to the broker or after power loss of the wifi_repeater. I have some other working code that does support this on my Sonoff switches.

martin-ger, 2017-09-23:
Might be a problem in the order of initialization - try the following in user_main.c (user_init() at the end). First registering the callbacks, than connecting to the broker:

`#ifdef MQTT_CLIENT
    mqtt_connected = false;
    mqtt_enabled = (os_strcmp(config.mqtt_host, "none") != 0);
    if (mqtt_enabled) {
	MQTT_InitConnection(&mqttClient, config.mqtt_host, config.mqtt_port, 0);

	MQTT_OnConnected(&mqttClient, mqttConnectedCb);
	MQTT_OnDisconnected(&mqttClient, mqttDisconnectedCb);
	MQTT_OnPublished(&mqttClient, mqttPublishedCb);
	MQTT_OnData(&mqttClient, mqttDataCb);

	if (os_strcmp(config.mqtt_user, "none") == 0) {
	  MQTT_InitClient(&mqttClient, config.mqtt_id, 0, 0, 120, 1);
	} else {
	  MQTT_InitClient(&mqttClient, config.mqtt_id, config.mqtt_user, config.mqtt_password, 120, 1);
	}
    }
#endif /* MQTT_CLIENT */
`

harryd100, 2017-09-23:
Tried it, no success, Pulling my hair out on this one. After MQTT is mqttConnectedCb, I see it subscribe to the right topic, but the retained message send by the broker is not seen by the mqttDataCb callback. I know the mechanism works because subsequent message arrive ok. It looks like the broker has already send the retained message before the the mqttDataCb callback is able to receive it.

harryd100, 2017-09-23:
Got it working, had to re-order the subscription to the various topics after MQTT connect to:
`static void ICACHE_FLASH_ATTR mqttConnectedCb(uint32_t *args)
{
	uint8_t ip_str[16];
	
	MQTT_Client* client = (MQTT_Client*)args;
	
	os_printf("MQTT: Connected\r\n");
	mqtt_connected = true;
#ifdef USER_GPIO_OUT
	if (os_strcmp(config.mqtt_gpio_out_topic, "none") != 0)
	{
		MQTT_Subscribe(client, config.mqtt_gpio_out_topic, 0);
		os_printf("Subscribed to: %s\r\n", config.mqtt_gpio_out_topic);
	}
#endif
	if (os_strcmp(config.mqtt_command_topic, "none") != 0)
	{
		MQTT_Subscribe(client, config.mqtt_command_topic, 0);
		os_printf("Subscribed to: %s\r\n", config.mqtt_command_topic);
	}
	os_sprintf(ip_str, IPSTR, IP2STR(&my_ip));
	mqtt_publish_str(MQTT_TOPIC_IP, "IP", ip_str);
}`

closed 2018-04-01 by martin-ger