martin-ger/esp_wifi_repeater #64: No AP after flashing NodeMCU

antonioag95, 2017-07-29:
After flashing my NodeMCU with the tool and with options as can be seen in the image, no AP shows up. I've tried several times also by using esptool.py from Windows but same result. Where am I wrong? Thank you for your hard work!
[image]

karawin, 2017-07-29:
Try DIO spi mode. Some non AI-THINKER esp need DIO only.

martin-ger, 2017-07-30:
Did you try the 1.5.4 firmware? What do you see on the serial port after reset?

antonioag95, 2017-07-30:
I also tried 1.5.4 firmware but it didn't work. When I reset it from serial I can see "csum err" and "ets_main.c". The vendor of non-working version of the board is "Hysiry" so there must be something different in this version. After several hours, I tried with another ESP8266 (there is no vendor written on it) and 1.5.4 is perfectly working on this board.

martin-ger, 2017-07-30:
Guess a problem with a faulty flash mem. Don't think it an issue of the software.

antonioag95, 2017-07-30:
I don't really think it is an hardware problem because the board is okay with Arduino IDE. Anyway, on a YouTube video which is related to your great software, someone commented saying he had to flash with the official NodeMCU Flasher in order to get it work but I don't really know the options there

karawin, 2017-07-30:
Do you hear me?
Try DIO spi mode. Some non AI-THINKER esp need DIO only.

Hysity needs the DIO spi mode.

Trust me, I encountered the problem for a user of my project.

antonioag95, 2017-07-30:
Thank you for your help, I can confirm that using DIO spi mode correctly flashed this board with 1.5.4 firmware (the other one didn't work). I suggest writing something about these particular boards in the readme file. Keep up the good work!

closed 2017-07-30 by antonioag95