martin-ger/esp_wifi_repeater #65: How to connect wifi open dhcp with portal user login password ?

ffanez, 2017-07-30:
Martin-ger's master,
Thank you for the repositories you have made.
I've managed to try it, thanks.
But can you help my problem to join the open dhcp wifi network using user login and password?
(Keep using the esp8266 device of course)

martin-ger, 2017-07-30:
Use the esp_wifi_repeater with a client. Log in with the client. From now on any further client doesn't need a new login.

ffanez, 2017-07-31:
Login and password not on the portal website page, but appear shortly after connecting on the wifi network if using on android. Sorry i do not understand its technical terminology.
Thanks.

RibeiroSantos, 2017-07-31:
- Config your esp_wifi_repeater to connect to the wifi network you want.
- config your esp_wifi_repeater to be an wifi network ("esp_wifi")
- config your client (in your case an android device) to connect to the new wifi network provided by esp_wifi_repeater ("esp_wifi")
- now, in your android, it will show the same as if you connect directly without using the esp_wifi_repeater

closed 2017-08-08 by martin-ger

ffanez, 2017-08-08:
Sorry it seems to use wpa2 enterprise.
Not the user login on the website page.
thanks.