martin-ger/esp_wifi_repeater #70: Read external ADC instead of internal

ulrikkau, 2017-08-07:
This is a super repeater. I have although a problem, to power it. With a solar panel and NiMh I occassionally run into that batteries are drained at cloudy weather, and I want to get the repeater in sleep mode at low power. I know it is implemented with with internal power, but as my solar power runs at 6V with a buckconverter I cannot use this feature, but it be wonderful if it can be run with external voltage on the AD pin instead.

martin-ger, 2017-08-07:
Interesting setup. Will go into that next week, when I am back from the beach... ;-)

geofrancis, 2017-08-08:
if your going to use an external pin can you make the voltage ratio configurable so that higher voltage packs can be read with a voltage divider.

ulrikkau, 2017-08-08:
Yes, actually I thought that I monitor one cell of the 4, ie. a 1- 1.5V, but still need a voltage divider to be able to measure within the 1V range of Esp8226. If I select two matched resistors with equal resistance ( around 50k) it will work well over a large temperature interval with standards resistors.
The setup is two battery powered esp8226 thermometer DS18b20 (running Esp easy) measuring the water temperature at the beach each ½ hour, and the signal is then going through the solar powered repeater situated on the roof of a shed to our house about 250 m away.

martin-ger, 2017-08-19:
I would add an option for reading the external ADC, but I do not fully understand how to access it. The doc on system_adc_read() says:
`• The 107th byte in esp_init_data_default.bin (0〜~127byte) is named as "vdd33_const",
and when wire TOUT pin to external circuitry, the vdd33_const must be set as real power
voltage of VDD3P3 pin 3 and 4.

• The range of operating voltage of ESP8266 is 1.8V~3.6V,the unit of vdd33_const is
0.1V,so effective value range of vdd33_const is [18, 36]. If vdd33_const is an ineffective
value which in [0, 18) or (36, 255), ESP8266 RF calibration will use 3.3V by default.
`
What does that mean to me? Do I have to set vdd33_const to something, before I get a useful value? TO what? 33? Any ideas or even better experiences?

geofrancis, 2017-09-06:
so what your saying is that the esp8266 would have to be run at a fixed voltage for the external ADC to use as a reference? then 3.3v would probably be the best to implement.

jjg, 2018-01-18:
I was just about to open an issue for this exact same feature, very happy to find I'm not the only one using esp_wifi_repeater with this setup.

I'm curious as to if any progress has been made?  I considered trying to make the change myself but I don't want to duplicate any existing efforts (also my C is pretty rusty so it might take me awhile).

gr01d, 2018-07-22:
I used an external code for adc_read() but I do not know if it works well.
http://41j.com/blog/2015/01/esp8266-analogue-input/

system_adc_read returns the maximum value each time I call it.

martin-ger, 2018-07-22:
Analog TOUT is controlled by the 107th byte of esp_init_data_default.bin (0 - 127 byte) is “vdd33_const“. It must be set to 0xFF i.e. 255 to read system voltage i.e. voltage on VDD pin of ESP8266. Init data is in config_flash.c (change this byte if you want to read the ADC):

Use this data:
```
const uint8_t esp_init_data_default[] = {
    "\x05\x08\x04\x02\x05\x05\x05\x02\x05\x00\x04\x05\x05\x04\x05\x05"
    "\x04\xFE\xFD\xFF\xF0\xF0\xF0\xE0\xE0\xE0\xE1\x0A\xFF\xFF\xF8\x00"
    "\xF8\xF8\x4E\x4A\x46\x40\x3C\x38\x00\x00\x01\x01\x02\x03\x04\x05"
    "\x01\x00\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00"
    "\xE1\x0A\x00\x00\x00\x00\x00\x00\x00\x00\x01\x93\x43\x00\x00\x00"
    "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
    "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
    "\x00\x00\x01\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"};
```