martin-ger/esp_wifi_repeater #74: Edit STA Hostname

klauscam, 2017-09-04:
Hi, First of all great work. I use this a lot on 4 different esp8266 with max download speed of around 8Mbps which is amazing.
Anyway just a suggestion, it would be great if we could change the hostname STA as well, instead of the NET_ssid :) thanks and good work

martin-ger, 2017-09-05:
Can be done, will add this in the next release. However, where do you see this name? In my environment actually nobody ssems to care about this string. At least I did not manage to see it in the fritzbox. That's why I more or less forgot this value...

klauscam, 2017-09-11:
Hi Thanks for getting back to me. So, this string is used as the hostname of the device. It is visible when I connect the device with my router, under list of stations there is the hostname, mac address and ip address.

BlueUpton, 2018-01-13:
I am another happy user of your project (hosted on a few wemos D1 mini pros) and would also like the ability to set the hostname for each esp for the same reason as klauscam. 👍

martin-ger, 2018-01-28:
Done: try "set sta_hostname <name>"

closed 2018-01-28 by martin-ger