martin-ger/esp_wifi_repeater #75: SSID with spaces and hotspots survey

imayoda, 2017-09-08:
Hi there, first of all congratulations for this incredible piece of software on a humble esp8266, very nice.
Only two questions: 
1. When I fill the SSID name with a space character within, it then outputs a SSID with a + plus character in place. Is it possible to fix or is it a C language constrain ?
2. Is it possible to implement a hotspot ap survey function in web interface? (I've seen an excellent function in [esp8266_deauther](https://github.com/wi-fi-analyzer/esp8266_deauther) here on github, with some minor caveats).
Kind regards

martin-ger, 2017-09-08:
You can enter a blank in the ssid (and any other string in the CLI) by using a backslash, e.g. 'My\ SSID' gives 'My SSID'.

That scan is somewhat more work. I'll thing about this for a future release.

imayoda, 2017-09-09:
Oh, great, it IS a special character.. thanks a lot for your answer and sorry for bothering as it was already written in docu. (shame on me :) )
Kind regards

closed 2017-09-09 by imayoda